SmingHub/Sming #282: feature/websockets clients

hreintke, 2015-09-13:
@hrsavla : @others
Some remarks from my side. I did not run the example but just reading your code.

- Naming
You define a callback WebSocketCompleteCallback but the function to set is called setOnDisconnectedCallback
Also in example application your naming is wsDisconnect
I would keep it consistent and everywhere use the "complete" variant.

- TCPconection receive
In onReceive you analyse the data received with assumption that a complete message is received.
I think the full message can come in multiple tcp fragments.
Beware I am not talking about websocket multi frame messages.

- onConnect vs onComplete
When WS handshake is unsuccessful you call this->connectedcallback(Mode);
With mode is wsDisconnect. Will onComplete also be called?     

- Example program
In the wsDisconnect when success = true you print 
"Websocket Client Disconnected Normally. End of program .."
when success = false 
"Websocket Client Disconnected. Reconnecting .."
But in both cases you  reconnect

alon24, 2016-03-14:
any news here? I now need websocket client

hreintke, 2016-05-23:
Discussion moved to SmingRTOS

closed 2016-05-23 by hreintke