SmingHub/Sming #283: si4432 compile error on MingGW

gernoteger, 2015-09-13:
# Problem
When compiling on my MinGW environment, I get this error:
~~~
C+ Libraries/si4432/si4432.cpp
In file included from Wiring/../include/user_config.h:27:0,
                 from Wiring/Arduino.h:6,
                 from Libraries/si4432/si4432.h:30,
                 from Libraries/si4432/si4432.cpp:27:
Libraries/si4432/si4432.cpp: In member function 'void Si4432::readAll()':
Libraries/si4432/si4432.cpp:400:98: error: iteration 7u invokes undefined behavior [-Werror=aggressive-loop-optimizations]
     (int ) allValues[i+12], (int ) allValues[i+13], (int ) allValues[i+14], (int ) allValues[i+15]
                                                                                                  ^
system/include/esp_systemapi.h:32:48: note: in definition of macro 'debugf'
 #define debugf(fmt, ...) m_printf(fmt"\r\n", ##__VA_ARGS__)
                                                ^
Libraries/si4432/si4432.cpp:394:21: note: containing loop
  for (byte i = 0; i < 0x7f; i+=16)
                     ^
cc1plus.exe: all warnings being treated as errors
make: *** [out/build/Libraries/si4432//si4432.o] Error 1
~~~

# Analysis

Code review shows, that indeed the last loop the last byte for debug output is taken from moutside the allocated buffer.
Since I it happens with stock settings (master+develop in Sming, Toolchain from cholatey install, I assume that this problem could happen to others.

# Environment
 gcc taken from Makefile database:
~~~
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -v
Using built-in specs.
COLLECT_GCC=c:\Espressif\xtensa-lx106-elf\bin\xtensa-lx106-elf-gcc.exe
COLLECT_LTO_WRAPPER=c:/espressif/xtensa-lx106-elf/bin/../libexec/gcc/xtensa-lx106-elf/5.1.0/lto-wrapper.exe
Target: xtensa-lx106-elf
Configured with: ../configure --prefix=/d/Neo/esp8266-devkit/Espressif/xtensa-lx106-elf --target=xtensa-lx106-elf --enable-multilib --disable-nls --disable-shared --disable-threads --with-gnu-as --with-gnu-ld --with-gmp=/d/Neo/esp8266-devkit/Espressif/build/gmp --with-mpfr=/d/Neo/esp8266-devkit/Espressif/build/mpfr --with-mpc=/d/Neo/esp8266-devkit/Espressif/build/mpc --enable-languages=c,c++ --with-newlib --disable-libssp --disable-__cxa_atexit
Thread model: single
gcc version 5.1.0 (GCC)
~~

gernoteger, 2015-09-13:
created PR #284 to solve this issue

ADiea, 2015-09-18:
nice catch

gernoteger, 2015-09-18:
UDK on Windows switched gcc from 4.8.2 to 5.1.0 between UDK 2.0.6 to UDK 2.0.7. You won't experience this issue on earlier UnofficialDevelopmentKits andprobably on Linux/Mac!.
Obviously loop optimizations changed between  those Versions

alonewolfx2, 2015-10-03:
It's solved with @gernoteger 's PR.

PR #293 (2015-09-19): Fix for #283 si4432 compile error on MingGW
this fixes #283:

# Solution
The fix just increases the buffer size by one. 

# Testing

I personally can't test the code at runtime for lack of this device. Tested compilation though.
closed 2015-10-03 by alonewolfx2