SmingHub/Sming #291: feature/rboot   non-rboot samples fail

robertkendrick, 2015-09-14:
SmingCore.h is included in all / most samples in Sming.

For the feature/rboot branch SmingCore.h includes Network/rBootHttpUpdate.h which in turn includes rboot-api.h

When making non-rboot sample ie using Makefile-project.mk,  complier complains about no such file rboot-api.h. Tried to fix this by adding the missing extra_includes that were present in the Makefile-rboot.mk file, but still complained about rboot-api.h not found. So added an extra_include $(SMING_HOME)/rboot/appcode in Makefile-project.mk.
This fixed the problem.
So it appears that the Makefile-project.mk requires:
EXTRA_INCDIR    ?= include $(SMING_HOME)/include $(SMING_HOME)/ $(SMING_HOME)/system/include $(SMING_HOME)/Wiring $(SMING_HOME)/Libraries $(SMING_HOME)/SmingCore $(SDK_BASE)/../include $(SMING_HOME)/rboot $(SMING_HOME)/rboot/appcode $(SMING_HOME)/rboot/user

I don't understand why Makefile-rboot.mk does not need EXTRA_INCDIR ?= ... rboot/appcode though.
Also all the sample top level Makefiles should have the    if RBOOT_ENABLED Makefile-rboot else Makefile-project      included I suppose.

robertkendrick, 2015-09-15:
Maybe the inclusion of Network/rBootHttpUpdate.h in SmingCore.h should be conditional on RBOOT_ENABLED being defined. 
Then you would not need any rboot includes in the 'project' makefile.

raburton, 2015-09-17:
This is already fixed, it just took a while to get merged into the feature/rboot branch.

closed 2015-09-28 by robotiko