SmingHub/Sming #294: Local override of user_init doesn't work

raburton, 2015-09-17:
user_main.cpp containing user_init() is compiled into both libsming.a and the user app. The version in the user app should take precedence over the libsming version, but it does not always (occasionally I have seen it do, my personal experience of trying to override functions in libraries with gcc has also shown this flakyness).
```
raburton@debian:~/dev/AnakodSming$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working directory clean
raburton@debian:~/dev/AnakodSming$ remote -v
bash: remote: command not found
raburton@debian:~/dev/AnakodSming$ git remote -v
origin	https://github.com/anakod/Sming.git (fetch)
origin	https://github.com/anakod/Sming.git (push)
raburton@debian:~/dev/AnakodSming$ git branch -v
  feature/rboot a5bbbed Merge pull request #279 from raburton/feature/rboot
* master        0bbdcc7 Merge pull request #273 from raburton/master
raburton@debian:~/dev/AnakodSming$ cd HttpServer_ConfigNetwork/
raburton@debian:~/dev/AnakodSming/HttpServer_ConfigNetwork$ make clean && make
C+ app/application.cpp
C+ /home/raburton/dev/Sming/Sming//appinit/user_main.cpp
AR out/build/app_app.a
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------
/bin/sh: 1: [[: not found

out/build/app.out:     file format elf32-xtensa-le

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000350  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       00002060  3ffe8350  3ffe8350  00000430  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00007568  3ffea3b0  3ffea3b0  00002490  2**4
                  ALLOC
  3 .irom0.text   0003b846  40209000  40209000  00007ba0  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .text         0000570d  40100000  40100000  00002490  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
------------------------------------------------------------------------------
# Generating image...
Generate firmware images successully in folder out/firmware.
Done
# Generating spiffs_bin
Checking for spiffs files
web/build directory exists. Creating spiff_rom.bin

raburton@debian:~/dev/AnakodSming/HttpServer_ConfigNetwork$ grep -B 1 user_init out/firmware/firmware.map 
/home/raburton/dev/Sming/Sming//compiler/lib//libsming.a(user_main.o)
                              /opt/esp-open-sdk//sdk/lib/libmain.a(app_main.o) (user_init)
```
Note the last couple of lines, showing that user_init is coming from from libsming, not app_app.a as it should.

raburton, 2015-09-19:
After further discussion in #268 there some suggestions for how to deal with this have been made. There is little point having user_init compiled in both libsming and the user app, when the one in the user_app doesn't get used. So the sming/appinit module should be removed from libsming Makefile or the apps Makefile-project.cpp / Makefile-rboot.cpp.

The advantage of removing it from libsming and leaving it in the app compile is that #defines can be specified per app, instead of having to be set when libsming is compiled. However the only thing that that is dependant on a #define in user_init is the option to disable spiffs auto-mounting (mounting here doesn't seem to be the appropriate place anyway).

The advantage of removing it from the app compile and leaving it in libsming is a simpler, neater app compile, and as it is unlikely anyone will need to change this anyway (except for the one case mentioned above) it is better just compiled once in libsming. To enable this spiffs_mount should be moved out of user_init and added to the apps init by the programmer if they wish to use spiffs with the auto-calculated length and position. If using rBoot the programmer will be expected to call spiffs_mount_manual from init anyway with appropriate parameters to set up the spiffs, so this change only affects those using the older method.

I propose the following change:
1) Remove `MODULES += $(SMING_HOME)/appinit` from Makefile-project.mk and Makefile-rboot.mk
2) Remove `spiffs_mount();` from user_init.cpp
3) Add `spiffs_mount();` to the init method of samples with actually use spiffs.

piperpilot, 2015-09-30:
+1 on this change as proposed by @raburton.  I think control of spiffs needs to be moved to the user app and leaving user_init.cpp where it is is appropriate.  The things that are done there are pretty much required for any app and should be initialized as standard.

hreintke, 2015-10-05:
@raburton @piperpilot 
If I recall correctly, the above is fixed in the rBoot implementation.
If I am correct please close this issue.

raburton, 2015-10-05:
No not at all. We work around it by getting the user to rebuild Sming with an extra #define set. This still needs fixing properly. I'll have a look at that tomorrow once I've finished my nights.

raburton, 2015-10-06:
PR done #319

hreintke, 2015-10-11:
@raburton : 
Can we close this issue or do we still have open issues on this ?

raburton, 2015-10-11:
we can close it, but it will close automatically when the pr is merged into master

hreintke, 2015-10-11:
ok, did not know that.
Learning on the job.

commit 686f7b3a (2015-10-08): Merge pull request #319 from raburton/spiffsmount

Remove appinit from user app modules list. Closes #294
commit 99671576 (2015-10-08): Remove appinit from user app modules list. Closes #294

The local override doesn't actually work. This meant you needed to recompile
libsming with a #define which then fixed the behaviour for all your apps.
User apps that wants to use spiffs now need to call spiffs_mount() in
their init method.
closed 2015-10-21 by raburton