SmingHub/Sming #295: Again, the problem with the system Debug Output

STASIUK, 2015-09-17:
problem with the system Debug Output!
Output system Debug information when disable 
>Serial.systemDebugOutput(false);

hreintke, 2015-09-17:
@STASIUK : 
Can you provide the sming and adk version you are using ?

Tavalin, 2015-09-17:
@STASIUK what's the actual issue?

STASIUK, 2015-09-19:
> scandone
add 0
aid 3
pm open phy_2,type:2 0 0
cnt 
> 
connected with myAPP, channel 10
dhcp client start...
ip:192.168.2.8,mask:255.255.255.0,gw:192.168.2.1

How to disable output this information?

flexiti, 2015-09-19:
http://bbs.espressif.com/viewtopic.php?f=13&t=610&sid=c12accff25fc0cf04b6f19e04f6a5cb0#unread

flexiti, 2015-09-19:
Read info from esspresif point 4:
4. it is the ROM log (startup log) we can't remove or disable it . But you could call system_uart_swap to use MTCK & MTDO as UART0 

Thanks for your interest in ESP8266 !

closed 2015-10-13 by STASIUK