SmingHub/Sming #297: millis() resets to zero after 71.58 minutes

larsch, 2015-09-25:
The 32-bit microsecond counter wraps to zero after 71.58 minutes, causing millis to do the same():

```c
unsigned long millis(void)
{
	return system_get_time() / 1000UL;
}
```

Tavalin, 2015-09-29:
This doesn't really sound like a Sming bug, as it from what I remember it originates in the Espressif SDK.

slaff, 2015-09-29:
Unfortunately that is what the hardware can provide. Definitely not a Sming bug.

piperpilot, 2015-09-29:
Closing as not a bug.  If you need to keep track of runtime, I suggest if you detect that the returned millis is less than the previous returned millis, that you add 32-bits to it and keep track of how many times you do that while the system is running.

larsch, 2015-09-29:
It's Sming doing the division. There's no millis() function in the SDK, only microseconds. Sming appears to be trying to emulate an Arduino function, but it's not compatible? Kinda hard to port over Arduino code that relies on millis().

piperpilot, 2015-09-29:
@larsch what are you looking for?  This relies on a system call that overflows... Are you saying that millis doesn't work at all so you can't implement your arduino code...

Help understand the problem and we can try to come up with a solution.

raburton, 2015-09-29:
I see the problem. Because the Sming version of millis is based on a 32bit microsecond counter it overflows 1000 times quicker than the arduino version. So the underlying issue with micros can't be fixed, that's an sdk limitation, but the millis function which is presumably there to make things more arduino compatible doesn't do the same as the arduino version.

larsch, 2015-09-29:
This code detects the wraparound and internally adds the missed 32 bits. It has one caveat, that it must be called once between every overflow in order to work. That could be done from a periodic timer, if not otherwise called often enough.

```c++
static uint32_t millis_offset = 0;
static uint32_t micros_extra = 0;
static uint32_t last_us = 0;

uint32_t millis() {
   uint32_t us = micros();
   if (us < last_us) {
      millis_offset += 4294967;
      micros_extra += 296;
   }
   last_us = us;
   return millis_offset + us / 1000 + ((us % 1000) + micros_extra) / 1000;
}
```

larsch, 2015-09-29:
I guess I'm just used to an uint32_t tick count overflow is quite normal, and even not a problem when calculating differences, since (millis() - lastMillis) will always give a useful result, given that you call it often enough, even when it overflows.

raburton, 2015-09-29:
Yes, I'd agree that the Sming version isn't really good enough. I'm not sure what the alternative is though. The time comes from the network device, maybe it keeps a longer count if we can work out how to access it. I'll have a poke around...

pvvx, 2015-10-18:
MAC_TIMER64BIT_COUNT: 0x3ff21048  * Step: 1us * Uses: wDev_MacTim1Arm() 
MAC_TIMER64BIT_COUNT_LW (*((volatile unsigned int *)(0x3ff21048)))
MAC_TIMER64BIT_COUNT_HI (*((volatile unsigned int *)(0x3ff2104c)))

linlun, 2015-11-05:
I have seen arduino programs written like this:
```tTimeToDoSomething = 	millis() + DELAY_TIME_MS;```
and then a check later like:
```
if (millis() >= tTimeToDoSomething)
{
//Do something
}
```
That event will never happen when the millis() function does not wrap together with the uint32 used as the variable.
My version of millis() that seems to work:
```
unsigned long millis(void)
{
	return (uint32)(micros_64() / 1000UL);
}
static uint64_t us_offset = 0ULL;
static uint32_t us_last = 0UL;
unsigned long long micros_64(void)
{
	uint32 now = system_get_time()+ 4200967295UL;
	if (us_last > now)
	{
		us_offset += ULONG_MAX;
	}
	us_last = now;
	return ((uint64_t)now+us_offset);
}
```
I think it is a bit easier to read, but it does use a 64 bit calculation (same amount of static data), also only one division and no modulus calculation.
You can also use the micros_64 if higher resolution is needed.
The same issue as with the version above, the micros_64() function must be called at least once every 72 minutes.

linlun, 2015-11-05:
The "+ 4200967295UL" should of course not be there, it was used to test the overflow without waiting for 72 minutes.

hreintke, 2016-03-25:
Closed due to no progress

closed 2016-03-25 by hreintke