SmingHub/Sming #299: Software serial? It's possible?

STASIUK, 2015-09-25:
Is it possible to run Software serial?

flexiti, 2015-09-25:
TX yes, RX will be more difficult -  It depends on the speed you want to get

STASIUK, 2015-09-25:
for example 9600
possible?

flexiti, 2015-09-25:
at that speed:
TX - (eg. timer) every 104 us
RX - one bit is about 104 us, there must be some sampling, eg. 10 times, it means that the polling (eg. the timer) would have to be every 10us (and can not be disturbed) to find start bit.
Or based on port interrupt on the edge and first timer after 52 us and next every 104 us)
Possible, but You must have a good knowledge of the system and interrupt priorities

eg. 110 baud - bit is 10 ms long  ,it is easier :)

hreintke, 2015-12-21:
No information for a long time.
Closing the issue

closed 2015-12-21 by hreintke

crosofg, 2017-04-25:
Is software serial possible? Can I connect four serial port devices?

flexiti, 2017-04-25:
Yes, as described above

crosofg, 2017-04-25:
Can some library like this [](https://github.com/plerup/espsoftwareserial) be used to connect four serial devices to esp8266?