SmingHub/Sming #303: Consider  cleanup AirUpdate example now that  rBoot example is in place.

robotiko, 2015-09-28:


alonewolfx2, 2015-09-28:
i think good idea. also airota seems not stable. @kireevco @anakod is it ok for you?

hreintke, 2015-09-28:
If we call the current master (including rboot) Sming Version 2.0.
Option would be to still add a small PR wich states a depreciation note to both AirUpdate and AirUpdate example noting the rboot as replacement. In release 2.1 -> completely remove AirUpate functionality

hreintke, 2015-10-30:
@raburton @kireevco : 
Still open question.
As rboot ota is now fully integrated into Sming 1.4.0 I proposeto remove the Airupdate and related code from Sming Core.

What is your idea on this ?

harry-boe, 2015-10-30:
Im new to the esp and to sming but i can confirm that the robot and the integration in sming – once you got used to it - works extremely well.

Therefore focusing one solution will help to keep the focus for new features to be added.
I consider to combine it with the smartconfig and maybe add some security to create a generic OTA bootstrap process.

BTW: thanks for all your efforts on swing, it’s by far the most advanced approach to build up ion the esp platform

hreintke, 2016-01-27:
Solved in Sming_RTOS.
Will not be backported to NONOS

closed 2016-01-27 by hreintke