SmingHub/Sming #304: Newbie question: what to flash where?

maiorfi, 2015-09-29:
Hi.

I set up both linux and windows environments.

Compiling Basic_blink I get an out folder with 3 .bin file, a couple of .map and .new files (on linux).

What should I flash where (in terms of memory regions)?

Btw, in Windows same files appear with different names (eagle.flash.bin, eagle.irom0text.bin and spiff_rom.bin)...what is going to be flashed where in this case?

Thanks!

piperpilot, 2015-09-29:
Hi @maiorfi.  Flashing is automatic if you run the make target "flash".  Depending on your OS, you will see it run a command similar to the following:

```
/opt/esp-open-sdk/esptool/esptool.py -p /dev/tty.usbserial -b 115200 write_flash -ff 40m -fm qio -fs 4m 0x00000 out/firmware/0x00000.bin 0x09000 out/firmware/0x09000.bin 0x38000 out/firmware/spiff_rom.bin
```

You will notice the compiled versions are in out/firmware and the 3 bin files are 0x00000.bin, 0x09000.bin and spiff_rom.bin.

closed 2015-09-29 by piperpilot

maiorfi, 2015-09-30:
Sorry to be a nuisance, but does such memory regions displacement depend on specific ESP hardware? I think I've got a ESP-12 with 512Kb Flash and programming it with the 3 images coming from BasicBlink demo doesn't seem to work as expected (i.e. led does not blink). In order to validate flashing parameters, I successfully flashed node_mcu image and it works.

Thanks :smile:

piperpilot, 2015-09-30:
Which ESP-12 do you have?  Does it have pins/pads down 2 sides or 3?  The ESP-12E (pads on 3 sides) needs an optional parameter of:

SPI_MODE = qio

Put that in your makefile-user.mk.

How are you determining the basic_blink isn't working?  With the node_mcu module, you won't see any blinking.  A better one to test with is basic_serial and connect up a serial terminal.

raburton, 2015-09-30:
Isn't it dio for the newer model (and qio is the default)?

piperpilot, 2015-09-30:
Ahhh shoot... you are right @raburton I was looking at the wrong makefile.  I did get this working with DIO and not QIO.

ESP-12E needs SPI_MODE = dio

maiorfi, 2015-09-30:
Ok, with DIO mode and using GPIO16 I got BasicBlink sample working (as well as BasicSerial one).

Thanks!