SmingHub/Sming #305: master, develop, future/rboot  question

flexiti, 2015-09-29:
I would like to take some new Sming version with rboot but I'm a little confused
master - has rboot
develop - no rboot ???
Futures / rboot - has

Which one to download?

alonewolfx2, 2015-09-29:
master has rboot now. its merged and recommended way.

raburton, 2015-09-29:
feature/rboot can probably be deleted now it has been merged, but no hurry as there might still be people using the branch. I guess rBoot needs to be merged back from master to develop. Going forward most big things will go through the develop branch, but little things going direct into master (or rBoot because it used a slightly older version of the new git development process currently in use) will have to be merged back.

closed 2015-09-29 by raburton

hreintke, 2015-09-29:
@raburton @alonewolfx2 : 
I am working with @robotiko to document the workflow we are going to use, it will be available on short notice. When that is accepted, we have to cleanup current branches. There are bugfixes and new features in different stadium of readiness in several branches.

raburton, 2015-09-29:
Ok, to the best of my knowledge there is no reason feature/rboot cannot now be deleted (everything is merged) so it's no problem if you want to clean this up (i.e. delete it).