SmingHub/Sming #313: Espressif's open lwip

alonewolfx2, 2015-10-01:
Can we change liblwip to Espressif's new open lwip? It would be very good and usefull.

pvvx, 2015-10-06:
SDK 1.4.0 (Open lwip: https://github.com/pvvx/MinEspSDKLib/tree/master/app/sdklib/lwip )
1) struct netif. netif.h. Add:
#if DEF_SDK_VERSION >= 1400 // (SDK 1.4.0)
  void * dhcpc_event; // +40 (SDK 1.4.0)
#endif
2) New eagle_lwip_if.c: https://github.com/pvvx/MinEspSDKLib/blob/master/app/sdklib/system/eagle_lwip_if.c

alonewolfx2, 2015-10-06:
@pvvx nice to see you :) i will try and let you know

pvvx, 2015-10-14:
Espressif: Open source LWIP for ESP_IOT_SDK_V1.4.0: http://bbs.espressif.com/viewtopic.php?f=46&t=1221

alonewolfx2, 2015-10-15:
@pvvx can you provide pr with esp open lwip for us ?

closed 2015-12-21 by hreintke