SmingHub/Sming #321: Long initialization time Sming after deep-sleep (SDK 1.3.0..1.4.0 error goto deep-sleep)

pvvx, 2015-10-06:
system_deep_sleep () finishes exception (system_get_rst_info () = REASON_EXCEPTION_RST instead REASON_DEEP_SLEEP_AWAKE)
Error in libmain.a: app_main.o
SDK 1.3.0..1.4.0 requires patch app_main.o.

alonewolfx2, 2015-10-06:
Can you provide patch information ?

pvvx, 2015-10-06:
Original version:
void system_deep_sleep_local_2(void)
{
	ets_intr_lock();
	Cache_Read_Disable();
	SPI0_CMD = 0x200000;
	while(SPI0_CMD);
	struct rst_info rst_info;
	ets_memset(rst_info, 0, sizeof(rst_info));
	rst_info.reason = REASON_DEEP_SLEEP_AWAKE;
	system_rtc_mem_write(0, &rst_info, sizeof(rst_info));
	IO_RTC_2 = 1<<20;  
}
Patch:
void __attribute__((noreturn)) system_deep_sleep_local_2(void)
{
	ets_intr_lock();
	Cache_Read_Disable();
	SPI0_CMD = 0x200000;
	while(SPI0_CMD);
	struct rst_info rst_info;
	ets_memset(rst_info, 0, sizeof(rst_info));
	rst_info.reason = REASON_DEEP_SLEEP_AWAKE;
	system_rtc_mem_write(0, &rst_info, sizeof(rst_info));
	IO_RTC_2 = 1<<20; 
	while(1);
}

08 -> 06
71 -> FF
12 -> FF

0008c5        	call0	40103a6c <system_rtc_mem_write>        
ffee61        	l32r	a6, 40103998 
fa5671        	l32r	a7, 4010233c 
0020c0        	memw                                           
426762        	s32i	a6, a7, 0x108                          
ffff06        	j	401039ec <---- patch

pvvx, 2015-10-06:
Maximum speed connection module (ST) to the outside of AP:

Set deep-sleep mode 1.

void ICACHE_FLASH_ATTR user_init(void) {
  ...
  wifi_set_sleep_type(0);
  wifi_station_dhcpc_stop(); // use static ip
  // Do not change the other configuration WiFi!
  ...  
  system_init_done_cb(init_done_cb);
}

void ICACHE_FLASH_ATTR init_done_cb(void) {
	...
	switch(system_get_rst_info()->reason) {
	case REASON_SOFT_RESTART:
	case REASON_DEEP_SLEEP_AWAKE:
		break;
	default:
		User_WiFi_config();
		break;
	}
	...
}

hreintke, 2015-10-06:
@pvvx @alonewolfx2 
What I read in the espressif BBS the REASON_DEEP_SLEEP_AWAKE issue is solved in SDK 1.4.0
http://bbs.espressif.com/viewtopic.php?f=7&p=3978#p3978
I would propose not to implement an intermediate solution but have it "solved" when we introduce SDK 1.4.0 support in Sming

alonewolfx2, 2015-10-07:
@hreintke thanks. i saw that. this will fix when sdk 1.4 update.

hreintke, 2015-12-21:
@alonewolfx2 : 
Can you confirm that SDK 1.4.0 solves the issue and if so close the issue ?

alonewolfx2, 2015-12-22:
It's solved with new sdk.

closed 2015-12-22 by alonewolfx2