SmingHub/Sming #322: sprintf vs os_sprintf

flexiti, 2015-10-06:
Can someone explain to me I doubt about the sprintf  vs  os_sprintf,   malloc   vs   os_malloc etc. in Sming?
@anakod @hreintke ? 
I noticed an impact on eg. TCP transmission (-8)

pvvx, 2015-10-06:
os_sprintf = define ets_sprintf -> SDK lib\libmain.a:eagle_lib.o ->  ets_vsprintf -> ets_vsnprintf

sprintf -> libmicroc.a -> https://github.com/anakod/esp_microc/blob/master/stdio/sprintf.c

hreintke, 2015-10-06:
@flexiti @pvvx : 
There has been a change to update the several printf function (printf, vsnprintf,...) to a local m_printf in order to add float support -> PR #188
The os_.. and ets_... functions do not provide that float support

Recently I saw an issue mentioning sprintf might not have correct float support but that has still to be invetigated

flexiti, 2015-10-08:
@pvvx @hreintke 
At the moment I close this topic.
I have a track, but unfortunately the test environment is complex and dependent on hardware - at least every 2-3 hours I have this famous error TCP -8
If you find something specific I will let you know in new topic

closed 2015-10-08 by flexiti