SmingHub/Sming #327: Memory leak while publishing large messages from an onReceived callback

dmarkey, 2015-10-08:
Repro steps:

Compile/Burn sample application.

Barrage queue with messages. like so: yes | mosquitto_pub -h dmarkey.com -p 8000 -t test -l

It will end with a crash:

realloc 13806 -> 13988
E:M 14008
Fatal exception 29(StoreProhibitedCause):
epc1=0x4023d258, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00003663, depc=0x00000000

dmarkey, 2015-10-08:
Sample app here: https://cpaste.org/puwj8jups

hreintke, 2015-10-08:
@dmarkey 
When I run your example my output is : 
```
connected with FRITZ!Box Fon WLAN 7360, channel 1
dhcp client start...
ip:10.0.0.217,mask:255.255.255.0,gw:10.0.0.138
I'm CONNECTED
MQTT start connection
connect to: dmarkey.com
timeout updating: 70 -> 65535
subscription 'test' registered
realloc 23 -> 163
DNS record found: dmarkey.com = 193.183.99.120
TcpConnection::connect result:, 0
OnConnected
TCP connected
onReadyToSendData: 0
TcpClient request completed
TCP sent: 34
onReadyToSendData: 2
> MQTT status: MQTT_MSG_CONNACK (len: 4)
TCP received: 4 bytes
onReadyToSendData: 1
> MQTT status: MQTT_MSG_SUBACK (len: 5)
TCP received: 5 bytes
onReadyToSendData: 1
TcpClient request completed
TCP sent: 2
onReadyToSendData: 2
> MQTT status: MQTT_MSG_PINGRESP (len: 2)
TCP received: 2 bytes
onReadyToSendData: 1
TcpClient request completed
TCP sent: 2
onReadyToSendData: 2
> MQTT status: MQTT_MSG_PINGRESP (len: 2)
TCP received: 2 bytes
onReadyToSendData: 1
TcpClient request completed
TCP sent: 2
onReadyToSendData: 2
> MQTT status: MQTT_MSG_PINGRESP (len: 2)
TCP received: 2 bytes
onReadyToSendData: 1
TcpClient request completed
TCP sent: 2
onReadyToSendData: 2
> MQTT status: MQTT_MSG_PINGRESP (len: 2)
TCP received: 2 bytes
onReadyToSendData: 1
TcpClient request completed
TCP sent: 2
onReadyToSendData: 2
> MQTT status: MQTT_MSG_PINGRESP (len: 2)
TCP received: 2 bytes
onReadyToSendData: 1
TcpClient request completed
TCP sent: 2
onReadyToSendData: 2
> MQTT status: MQTT_MSG_PINGRESP (len: 2)
TCP received: 2 bytes
onReadyToSendData: 1
TcpClient request completed
TCP sent: 2
onReadyToSendData: 2
> MQTT status: MQTT_MSG_PINGRESP (len: 2)
TCP received: 2 bytes
onReadyToSendData: 1
TcpClient request completed
TCP sent: 2
onReadyToSendData: 2
> MQTT status: MQTT_MSG_PINGRESP (len: 2)
TCP received: 2 bytes
onReadyToSendData: 1
TcpClient request completed
TCP sent: 2
onReadyToSendData: 2
> MQTT status: MQTT_MSG_PINGRESP (len: 2)
TCP received: 2 bytes
onReadyToSendData: 1
TcpClient request completed
TCP sent: 2
```

dmarkey, 2015-10-08:
Are sending a barrage of messages using 'yes | mosquitto_pub -h dmarkey.com -p 8000 -t test -l'

hreintke, 2015-10-08:
Sorry, do not understand what you are saying.

dmarkey, 2015-10-08:
yes | mosquitto_pub -h dmarkey.com -p 8000 -t test -l 

That's a bash command. Just make sure you have mosquito installed. It sends lots of messages.

hreintke, 2015-10-08:
OK, have work on that first. I am not that familiar with MQTT, only did some easy testing some time ago.
I am on windows, was using mqtt-spy then. But not sure whether I can push a lot of messages with that

dmarkey, 2015-10-08:
I can run it when I get home.

dmarkey, 2015-10-08:
You should be seeing the symptom now.

hreintke, 2015-10-08:
Sorry, cannot test now. When I am testing later (will be tomorrow) will the messages still be in ?
Is there a way that you can adapt your example that by sending/publish a first message the burst is created automatic ?

dmarkey, 2015-10-09:
Sorry, ill write a small script to burst the messages.

dmarkey, 2015-10-15:
Any progress then :)

hreintke, 2015-10-16:
Because I did not hear that you finished the script to burst the message.

dmarkey, 2015-10-16:
It's running now. I'll leave it running indefinitely.

hreintke, 2015-10-16:
Just have the same output as above.
How do you start the burst ? A

dmarkey, 2015-10-16:
Sorry, my script was pointing at the wrong server. It's running now

hreintke, 2015-10-16:
Results now : 

My environment : Nodemcu with esp12e, latest sming release.

```
connected with FRITZ!Box Fon WLAN 7360, channel 1
dhcp client start...
ip:10.0.0.217,mask:255.255.255.0,gw:10.0.0.138
I'm CONNECTED
MQTT start connection
connect to: dmarkey.com
timeout updating: 70 -> 65535
subscription 'test' registered
realloc 23 -> 163
DNS record found: dmarkey.com = 193.183.99.120
TcpConnection::connect result:, 0
OnConnected
TCP connected
onReadyToSendData: 0
TcpClient request completed
TCP sent: 34
onReadyToSendData: 2
> MQTT status: MQTT_MSG_CONNACK (len: 4)
TCP received: 4 bytes
onReadyToSendData: 1
> MQTT status: MQTT_MSG_SUBACK (len: 5)
> MQTT status: MQTT_MSG_PUBLISH (len: 42)
4: 34

test:
        {"name": "switch", "switchnum": 1}
> MQTT status: MQTT_MSG_PUBLISH (len: 42)
4: 34

test:
        {"name": "switch", "switchnum": 1}
realloc 117 -> 363
> MQTT status: MQTT_MSG_PUBLISH (len: 42)
4: 34

test:
        {"name": "switch", "switchnum": 1}
> MQTT status: MQTT_MSG_PUBLISH (len: 42)
4: 34

test:
        {"name": "switch", "switchnum": 1}
realloc 351 -> 533
> MQTT status: MQTT_MSG_PUBLISH (len: 42)
4: 34

```
More of the same ....
```
test:
        {"name": "switch", "switchnum": 1}
realloc 7488 -> 7670
> MQTT status: MQTT_MSG_PUBLISH (len: 42)
4: 34

test:
        {"name": "switch", "switchnum": 1}
realloc 7605 -> 7787
> MQTT status: MQTT_MSG_PUBLISH (len: 42)
4: 34

test:
        {"name": "switch", "switchnum": 1}
realloc 7722 -> 7904
> MQTT status: MQTT_MSG_PUBLISH (len: 42)
4: 34

test:
        {"name": "switch", "switchnum": 1}
realloc 7839 -> 8021
> MQTT status: MQTT_MSG_PUBLISH (len: 50)
TCP received: 1452 bytes
> MQTT status: MQTT_MSG_PUBCOMP (len: 36)
> MQTT status: b0 (len: 6)
> MQTT status: MQTT_MSG_PUBCOMP (len: 36)
> MQTT status: b0 (len: 6)
> MQTT status: MQTT_MSG_PUBCOMP (len: 36)
> MQTT status: b0 (len: 6)
> MQTT status: MQTT_MSG_PUBCOMP (len: 36)
> MQTT status: b0 (len: 6)
> MQTT status: MQTT_MSG_PUBCOMP (len: 36)
> MQTT status: b0 (len: 6)
> MQTT status: MQTT_MSG_PUBCOMP (len: 36)
> MQTT status: b0 (len: 6)
> MQTT status: MQTT_MSG_PUBCOMP (len: 36)
> MQTT status: b0 (len: 6)
> MQTT status: MQTT_MSG_PUBCOMP (len: 36)
> MQTT status: b0 (len: 6)
> MQTT status: MQTT_MSG_PUBCOMP (len: 36)
> MQTT status: b0 (len: 6)
> MQTT status: MQTT_MSG_PUBCOMP (len: 36)
> MQTT status: b0 (len: 6)
> MQTT status: MQTT_MSG_PUBCOMP (len: 36)
> MQTT status: b0 (len: 6)
> MQTT status: MQTT_MSG_PUBCOMP (len: 36)
> MQTT status: b0 (len: 6)
> MQTT status: MQTT_MSG_PUBCOMP (len: 36)
> MQTT status: b0 (len: 6)
> MQTT status: MQTT_MSG_PUBCOMP (len: 36)
> MQTT status: b0 (len: 6)
```
More of the same : 

No crash....

I do see the increase of the realloc but would like to see the crash for easier troubleshooting.

hreintke, 2016-03-25:
Closed due to no progress

closed 2016-03-25 by hreintke