SmingHub/Sming #328: Feedback on Sming Release v1.3.0

hreintke, 2015-10-08:
@All : 
I just prepared a PreRelease Sming v1.3.0 on Github.
Please comment on the terminolgy/content of the release. 
Actual functional/bug comments should be reported in separate issues.

If no or only small remarks are coming I will replace the PreRelease by Release tag and post a forum message.

raburton, 2015-10-08:
I've added a couple of links to upstreams for included components, but I don't know them all. Hope that's ok.

raburton, 2015-10-08:
Btw, is the double dot deliberate while it's pre-release or is it just a typo?

hreintke, 2015-10-08:
typo : fixed

hreintke, 2015-10-08:
and of course no problem updating, that's why the question is here

raburton, 2015-10-09:
is it too late to squeeze this [fix](https://github.com/SmingHub/Sming/pull/293/files) in? Apparently it breaks the build on windows with gcc 5.1

hreintke, 2015-10-10:
@raburton : 
We should ask our self the question : 
- Who would need this patch NOW instead of next release
- Is there a workaround for the patch available
- Is the patch tested and OK ?
- Is the patch tested on Backward compatibility

MY answers, anyone to comment
- Users of UDK_208/xtensa-lx106-elf-gcc.exe 5.1.0 which have si4432
- Use of Develop branch of sming
- ?
- ?

Based on MY answers I would recommend not to squeeze in current Sming Version

hreintke, 2015-10-10:
@raburton @others
Should add a section "External Dependencies" ? 
- Espressif SDK v1.3.0
- UDK 2.0.7  -----> include ?
- Xtensa toolkit v?? -----> include ? or just xtensa gcc version
- rBoot v??
- esptool2 v??
- Spiffy v??
- Spiffs v??

This is from my "Windows view" 
other maybe "esp_open_sdk" ??

kireevco, 2015-10-10:
@hreintke maybe for now. Eventually I'm planning to add some sort of installer that will pull down all binary dependendices.

Also, `choco install sming -y -source 'https://www.myget.org/F/kireevco-chocolatey/'` right now takes care most of things for windows, we just need to keep it up-to-date.

raburton, 2015-10-10:
@hreintke fair enough. Lets get this one released then and we can work on the next version, with the fixes in it. But to answer your questions:

>Who would need this patch NOW instead of next release

Anyone with gcc 5.1 (the library that that fails to build is part of libsming, so it doesn't matter if they actually want to use it or not).

> Is there a workaround for the patch available

No workaround, the only thing you can do to fix it is to apply the patch to your local copy.

> Is the patch tested and OK ?

Yes.

> Is the patch tested on Backward compatibility

Yes.

alonewolfx2, 2015-10-11:
> I know, @alonewolfx2 loves GUI :)

almost every windows users like GUI :)

hreintke, 2015-10-11:
And we should allow "selectable installation" 
Only Core,
Core + sdk
Core + spifyy + ... + ...
Core version v1.x.0, version v1.y.o

And : It should be fully located on an "open environment", not using private storage.

PS : I don't know whether the current choco/docker meets those requirements, used only manual installation and upgrade. :smile:

kireevco, 2015-10-11:
Choco somewhat does, but it is not crossllarform

kireevco, 2015-10-11:
I like gui too, but I know cli gives me rich automation ability :)

raburton, 2015-10-11:
added a note to the release notes reminding people the libsming.a binary is not included and needs to be built
can we remove the pre-release tag yet?

hreintke, 2015-10-11:
@raburton : @alonewolfx2 
Just removed the pre-release tag.
When @alonewolfx2 adds the "News and Releases" subforum on esp8266 I will announce it there too.

closed 2015-10-11 by hreintke