SmingHub/Sming #337: How to configure MQTT for LWT?

ashubhatt, 2015-10-13:
I am trying to configure my ESP to post a message with Last Will and Testament (LWT) on MQTT. I cannot find any example or any function declared for the same in the header file.

Can anybody point me in right direction?

slaff, 2015-10-14:
Take a look at my recent PR: https://github.com/slaff/Sming/commit/23997249620d9173c1ee29b21db41979a3cf015f. Once it is in the master branch, if it makes it to there, you should be able to easily set LWT.

kireevco, 2015-10-30:
@ashubhatt Do you think this solves your issue?

hreintke, 2015-11-14:
Solved by #390

closed 2015-11-14 by hreintke

ashubhatt, 2015-12-17:
I dont think the LWT feature works in Sming.

I tried to check the working, but after disconnection, the LWT message is not sent.
I am using the latest build - 1.4.0