SmingHub/Sming #339: Possible memory leak at ArduinoJson

vsirvent, 2015-10-13:
Hi,

I'm experiencing memory leaks when using JsonObject::addCopy(const String& stringKey, const String &stringVal).

After some inspection, it seems that JsonBuffer::createStringStorage is called, where new (this)JsonStringStorage is created. It uses overloaded operator new at JsonBufferAllocated which puts the new object in its own memory. The thing is that JsonStringStorage contains a String that internally allocs dynamic memory to store the string (String::buffer). This buffer is never deleted, as JsonStringStorage is created using JsonBufferAllocated and its destructor is never called. I think that JsonBufferAllocated derived classes cannot contain dynamic memory allocation, as it's never freed.

Do I miss anything?

Regards,

Vicente Sirvent

vsirvent, 2015-10-14:
Well, I can confirm the memory leak. Any class derived from JsonBufferAllocated that dynamically allocs memory will cause a memory leak (like JsonStringStorage). To avoid the memory leak you can avoid dynamic memory allocation in the class or keep track of the allocated classes in JsonBuffer and delete them in the destructor of JsonBuffer.

hreintke, 2015-10-15:
@piperpilot @robotiko : 
You have been looking in a "next version" on arduinojson. 
Can you check whether that has been fixed in there ?

hreintke, 2015-11-12:
Solved by #402

closed 2015-11-12 by hreintke