SmingHub/Sming #341: Binary files in the distribution package

kireevco, 2015-10-14:
Hi all,
Let's sync up on the topic of how do we distribute sming for all users.
I see it this way: 
1. We build sming in the CI, build binaries (currently libsming and spiffy) for all platforms. This will be available as sming-core or some similar zip file. We will have 3 files as a result of this build:
- sming-core-1.0.4-linux-x86_64.tar.gz 
- sming-core-1.0.4-win32-x86_64.zip 
- sming-core-1.0.4-darwin-x86_64.zip

2. We package a full isolated environment that contains a specific sdk version (tested one), and other external tools. This package contains sming-core contents as well.
- sming-1.0.4-linux-x86_64.tar.gz 
- sming-1.0.4-win32-x86_64.zip 
- sming-1.0.4-darwin-x86_64.zip

I will update this proposal when I get more details, but until then feel free to comment.

kireevco, 2015-10-14:
Good news, I was able to run OSX on my Proxmox cluster. This will allow use build on mac too:
[image]

raburton, 2015-10-14:
Quick thought, will write it while I think about it. We need to add a "release" or similar make target. It will build everything and then call a limited clean target that removes all the intermediate files but just leaves the binaries we want (where clean normally cleans everything). Then a new version can simply be checked out, built with release target and then zipped.

hreintke, 2015-10-14:
Some thoughts on timing.
Now/soon we have the major update on rboot integration/spiffy/spiffs and a number of fixes ready.
I think it would be wise to take this as a Sming release 1.4.0 before adding more functionality.

Question the is whether we would release 1.4.0 in the "old way" or are already (partly ?) ready for taking (parts of) the ideas above ?

kireevco, 2015-10-14:
I am not ready to make a cross platform build yet.
--
Sincerely,
  Dmitry Kireev
  Cell: +1 (818) 850-55-77
  Skype: kireev.co <http://kireev.co/>

raburton, 2015-10-14:
@hreintke rBoot was integrated in v1.3.0. We do have plenty of new stuff though, and bug fixes, waiting to go. I'd like the new spiffs/spiffy to get a more testing before it goes into a release though. If there are no more issues tomorrow (as people are currently testing with it) I'll merge it to develop so it gets wider testing. I suggest development freeze around that time and assuming no blockers within a couple of days release v1.4.0, in the same was as v1.3.0. This issue is going to take a little time to get up and running so we shouldn't wait in the mean time.

raburton, 2015-10-14:
@kireevco do you realise that little image you posted is 3.8mb? Not very mobile friendly!

kireevco, 2015-10-14:
@raburton ah... stupid retina screenshots. sorry.

hreintke, 2015-10-15:
@raburton : 
Yes, that was my idea too.
Once we freeze I will submit an issue "Feedback on Release 1.4.0" where we can collect issues/opinions on the release to come.
One thing one forehand : we will need documentation as there are actions needed from users to upgrade without problems.

kireevco, 2015-10-19:
Moving along:

[image]

kireevco, 2015-12-13:
I got esp-alt-sdk built on my cluster, all platforms work correctly. I'll work on sming build steps. 
Good news, that travisci has mac builder now!

hreintke, 2016-03-25:
Travis running implemented.

closed 2016-03-25 by hreintke