SmingHub/Sming #353: odd problem with develop branch and HttpServer_AJAX example

avr39-ripe, 2015-10-16:
Clone fresh copy of Sming from git, git checkout develop, make rebuild, make spiffy, make HttpServer_AJAX example and flash it into ESP-07. Only change SSID & PWD in application.cpp and change to use gcc48 in spiffy Makefile. ESPSDK 1.3.0 and also try 1.40 - same result. Platform of build - FreeBSD.

Then when I try to get some pages from esp-07 I get odd things on the serial and no data in browser.
What I doing wrong?? How to fix it?

Serial log:

timeout updating: 70 -> 65535
Server timeout updating: 40 -> 90
+TCP connection
timeout updating: 70 -> 65535
Server timeout updating: 40 -> 900
fs.start: size:208 Kb, offset:0x0x48000

mount res: 0

Station configuration was updated to: infjust
mode : sta(18:fe:34:9d:18:5e)
add if0
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with infjust, channel 1
dhcp client start...
ip:10.2.113.126,mask:255.255.255.128,gw:10.2.113.1
I'm CONNECTED
addUser: me 123
'/' registered
'/ajax/input' registered
'/ajax/frequency' registered

=== WEB SERVER STARTED ===
10.2.113.126
==============================

pm open phy_2,type:2 0 0
onAccept state: 0 K=0
Free heap size=38424, K=0
+TCP connection
timeout updating: 70 -> 90
TcpServer onClient  10.2.113.100, activeClients = 1
 
path=/ajax/input/
Host === 10.2.113.126
parsed
Request: GET, nodata
TCP received: 417 bytes
onReadyToSendData: 1
response sendHeader
response sendBody
realloc 1 -> 131
Stream completed
TCP connection closing
~TCP connection
-TCP connection
onAccept state: 0 K=0
Free heap size=37608, K=0
+TCP connection
timeout updating: 70 -> 90
TcpServer onClient  10.2.113.100, activeClients = 2
 
onAccept state: 0 K=1
Free heap size=37528, K=1
+TCP connection
timeout updating: 70 -> 90
[... 325 lines cut ...]
plain template text pos: 0, len: 1024
READ Template (0)
plain template text pos: 0, len: 1024
READ Template (0)
plain template text pos: 0, len: 872
TCP sent: 2920
onReadyToSendData: 2
response sendBody
READ Template (0)
plain template text pos: 0, len: 1024
READ Template (0)
plain template text pos: 0, len: 1024
READ Template (0)
plain template text pos: 0, len: 872
TCP sent: 2920
onReadyToSendData: 2
response sendBody
READ Template (0)
plain template text pos: 0, len: 1024
READ Template (0)
plain template text pos: 0, len: 1024
READ Template (0)
plain template text pos: 0, len: 872
TCP sent: 2920
onReadyToSendData: 2

raburton, 2015-10-16:
what version of spiffy are you using?

avr39-ripe, 2015-10-16:
Hello! I use spiffy from Sming itself. Sming AFTER merge of feature/newspiffy into develop. Use develop branch of Sming. Try SDK 1.3.0 and 1.4.0. esp-open-sdk as a compiler. Dev env is fully working, on older Sming with spiffy from external repo all works just fine. So help please :)

raburton, 2015-10-16:
Ok, you have matched versions then so that's good.
I don't know what the log is showing and if it's right or not, do you have one from when it worked? Something I notice though is that the template test doesn't go up, yet it seems to be able to keep read 3 chunks that add up to the total data sent, looks like it's working but not advancing the position. Although without seeing a working version I don't know if that's what it looked like before.

raburton, 2015-10-16:
I'm at work so can't test this, but I might have found the problem.
Change line 122 in DataSourceStream.cpp from:
`bool result = fileSeek(handle, len, eSO_CurrentPos) == 0;`
to:
`bool result = fileSeek(handle, len, eSO_CurrentPos) >= 0;`
And rebuild libsming.

raburton, 2015-10-17:
@avr39-ripe did that work?

avr39-ripe, 2015-10-17:
can't check yet.. sorry. will check it out asap! THANK YOU a lot anyway!!!

raburton, 2015-10-17:
Not to worry, I haven't had chance to test it either, been on call all day yesterday and on now for whole weekend. But luckily did get a little gap to look at the code yesterday. It's definitely a bug, the return code for the spiffy call has changed and the code needs to be updated, but it would be nice to confirm that it also fixes your bug before I commit it.

avr39-ripe, 2015-10-19:
@raburton Thank you a lot for you fix! As I see you already merge it to develop, try it and it WORKS like a charm! So #358 fix this issue.

closed 2015-10-19 by avr39-ripe

raburton, 2015-10-19:
That's great, thanks for the feedback.