SmingHub/Sming #355: Development freeze for v1.4.0

raburton, 2015-10-18:
I propose a development freeze in preparation for v1.4.0. Is there anything else out there (PRs) that have been tested and are ready to go into develop but haven't yet been? I'm not actually starting the development freeze yet, just proposing it, so there is still a window to put more in. If there is general agreement then we can officially start the freeze in a day or two for a short period of final testing of the complete develop branch. Important bug fixes only after that point. Development in PRs can continue during that time, just not be merged until after the next release.

slaff, 2015-10-18:
@raburton May be you can include #349 and #350 (if it is working for you too).

hreintke, 2015-10-18:
@raburton : 
Agree on freezing and releasing.
- I'd like to have #199 still in 
- In working with ftp on rboot I found a an issue in FTP.
Needs a small fix. -> will create a PR on monday (CET)
- I will create a PR to have the right Sming version included (monday)
- We do need description on how to upgrade from 1.3.0 -> 1.4.0

hreintke, 2015-10-18:
On #349 and #350 
We should not "push PR's" because of releasing.
Just codereview is not enough to get it into a release.

I propose to leave both to Release 1.5.0

raburton, 2015-10-18:
Agreed, doing a quick code review on #349 & #350 (while my on-call shift is not too busy) and found a few issues, but I'm not really going to have much chance to test these out anyway. We need to get a bit more wider feedback on them in the PRs before we can merge them into develop. Don't think they'll make it into v1.4 but hopefully into v1.5 because they sounds good in principal.

Is #199 ok to merge to develop?

raburton, 2015-10-18:
> We do need description on how to upgrade from 1.3.0 -> 1.4.0

We need a note about the change to spiffs requiring an updated spiffy, and the fact that that is now included and how to build it. Is there anything else?

hreintke, 2015-10-18:
Yes #199 is ok. It is a textual change to a debug message.

hreintke, 2015-10-19:
PR for Sming version submitted
I'll leave the FTPupdate for next release, want to do some more testing.
Do I need to take additional action so that #199 can be merged ?

Documentation.
I think we should have a wiki page on spiffs/spiffy and one on rBoot.
Also needing Release Notes. Would that be on Wiki/Release page/both ?

hreintke, 2015-10-19:
@kireevco : 
What is the status of the Wiki page of Windows/Linux/MaxOS Quickstart ?
Is the choco (?) environment still up to date ?

raburton, 2015-10-19:
Created a pr for the merge of develop into master #360, I'll keep it open throughout the development freeze so anyting else that goes into develop will get picked up by it too.

raburton, 2015-10-21:
v1.4.0 merged, just awaiting release...

closed 2015-10-21 by raburton