SmingHub/Sming #356: fast access to time (SystemClockClass::now)

johndoe8967, 2015-10-18:
It takes quite some time to access the time and convert it to unix time_t.
There could be a faster and direct way...

time_t SystemClockClass::nowUnix(TimeZone timeType /*= eTZ_Local*/) {
	uint32_t systemTime = RTC.getRtcSeconds();
	if ((timeType == eTZ_Local) || (status == eSCS_Initial))
	{
		return systemTime; // local time
	}
	else
	{
		return (systemTime - (timezoneDiff * SECS_PER_HOUR)); // utc time
	}
}

Tavalin, 2015-10-19:
Could you elaborate a bit on your use case/issue?

johndoe8967, 2015-10-20:
UseCase: SmartMeter
I have an interrupt function for each energy pulse which needs to store a timestamp.

The timestamp should be quite accurate so I use the micros() function but I also need a date/time so I also store the time_t for further processing

Tavalin, 2015-10-20:
The granularity of the SystemClock / Unix time is in seconds.  Are you saying that this method takes longer than a second to return?

johndoe8967, 2015-10-22:
no, no, 
I'm just always taking care about the cpu time of an interrupt routine :-)

hreintke, 2016-01-27:
Closed due to low activity

closed 2016-01-27 by hreintke