SmingHub/Sming #357: Transimssion TCP -8 error

flexiti, 2015-10-18:
Can anyone verify this?
@hreintke  ?

https://github.com/flexiti/flexiti_test

TCP transmision error -8

If you receive the first error, the error remains!!!!! until we make reset

Httcp client stops working, while tcp client seems to be resistant to this error

The physical display is not needed for the test, only SPI transfers

In my tests error occurs between 15-60 minutes and already then stays until reset

During the test, you should pay attention to the limitations thingspeak
  - sending not more often than every 15 seconds.

 (Nor do I know how some people will simultaneously send the same KEY.
   - Preferably make yourself your own key.

 Here is KEY=7XXUJWCWYTMXKN3L from Siming example

raburton, 2015-10-18:
I saw the same with `rBootHttpUpdate`, which is a subclass of `HttpClient`, had to create a new object each run to be sure it was clean. That also meant the object didn't exist unless it was needed at least once, so it didn't need storage allocating if it wasn't going to be used. Obviously some state is not getting cleared after the error. Try calling `reset` on the object and see if that fixes it.

flexiti, 2015-10-18:
I wrote a test program to detect the reasons for this error, and you're telling me to modify the test program to hide it ? :)
reset is not an option, I need to find the cause
Firstly I would like to know whether the test program works with similar FAULTS, in others users,
and next find out:
 - what "is not cleared" when an error occurs
- Why is it such a big impact  (printf or SPI or something else) on tcp transmission

raburton, 2015-10-18:
The reset method, if it works, isn't hiding the error it would be resetting it. Why do you think that resetting a failed connection before trying to use it again is unreasonable? Also, you can easily see what the reset method does, so if you call it and it helps then you have narrowed down the status bits that you need you look at. On the other hand I have now lost interest in your issue and will not be looking at any it further.

hreintke, 2015-10-19:
@raburton : @flexiti 
Let's keep communicating without getting angry.
If i read the above there is misunderstanding between reset an restart !

@flexiti : If the same error occurred in Richards rHttp it might lead to a cause/solution
@raburton : I agree with flexiti that we should try to get a cause and not a workaround.

But... Only with combined effort we will get Sming complete and stable

avr39-ripe, 2016-01-14:
Hello, I hit the same issue while trying to communicate between two esp, one as httpServer and other as httpClient. Sometimes I get this "magic error" and only reset device helps. Is there some better solution?? @flexiti did you came to something new with thi issue?

flexiti, 2016-01-14:
@avr39-ripe 
Can I see Your program ?
I have 8 ESP connected to the server and works me pretty well now.
I avoided all Delays and did change all functions that use the timer class.
https://thingspeak.com/channels/26576

avr39-ripe, 2016-01-15:
@flexiti 
Indeed yo can see the code! :)
Here is server side: https://github.com/avr39-ripe/TyTherm/tree/sensor_base
And here is client: https://github.com/avr39-ripe/OctoTherm/tree/http_stability
I use it in combination with esp 1.5.0 sdk + not very old Sming (I think something pre last release of sming)
There is NO delay() in client app. what do you mean by "change all functions that use timer class?" I use 4 seconds timer to update temperature form server.. I think there is no problem with server, both its web interface or curl from console CAN get http responses, even server app can, but NOT reliable.. If it works, it works for long time, but if at startup client get error -8 it stall forever till device reset.

avr39-ripe, 2016-01-15:
Looking to log more accurately and find out, that error is: 

"Download: http://10.2.113.114:80/state
connect to: 10.2.113.114
TcpConnection::connect result:, -4"

And without error condition - "TcpConnection::connect result:, -0"

So can anybody tell me what can cause randomly occurred error -4 ??

flexiti, 2016-01-15:
#define ESPCONN_RTE        -4    /* Routing problem.         */

flexiti, 2016-01-15:
@avr39-ripe 
I look at the code in the evening

avr39-ripe, 2016-01-15:
@flexiti already found that "-4" is "EHOSTUNREACH,  /* ERR_RTE        -4      Routing problem.         */"
But what can cause such "problem" or host unreachability in constant networking env? :(
Ok, thanks! will wait your feedback :)

avr39-ripe, 2016-01-15:
Some update:
After changing WIFI configuration from dhcp to static IP all become reliable and stable, no more -4 error, so something wrong is within my networking ;( can't figure out what. Other devices getting their IP by dhcp works just fine.

closed 2016-03-25 by hreintke