SmingHub/Sming #361: Notice SPI flash errors

hreintke, 2015-10-20:
@raburton : 
To notice SPI flash errors we could add a read after write to the 
`bool ICACHE_FLASH_ATTR rboot_write_flash(rboot_write_status *status, uint8 *data, uint16 len)`
in the rboot-api.c

Just a quick pseudo code : 
```
		// write current chunk
		//os_printf("write addr: 0x%08x, len: 0x%04x\r\n", status->start_addr, len);
		if (spi_flash_write(status->start_addr, (uint32 *)buffer, len) == SPI_FLASH_RESULT_OK) {
			//pseudo code to be detailed 
			// uint_8 check_buffer*
			//	check_buffer = (uint8 *)os_zalloc(len));
			// if (!check_buffer) {
					//os_printf("No ram!\r\n");
			//			return false;
			//	}
			//spi_flash_read(status->start_addr, (uint32 *)check_buffer, len)
			//if compare (buffer to checkbuffer != OK ) ret false;

			ret = true;
			status->start_addr += len;
		}
```
What is your opinion about this ?

hreintke, 2015-10-20:
@raburton : 
I am looking into an rBootControl class in which I will "hide" a number of rboot details to give an east/high level interface for users/developers.
For that some questions about the rBoot implementation.

In rbootconfig you only have the starting address of a rom, would it be good to also have the length so the update process can prevent overwriting next roms ?

In the example you use the uint_32 roms[..] addreses for "application location" and application supplied starting addresses for spiffs. 

Is there anything preventing that the roms[...] contains addresses and lengths of partitions of the full rom which can be populated with either of them ?

raburton, 2015-10-21:
Suggestion one (write verification) could easily be added, but it will double the amount of memory required to perform the complete operation. I will have a think about that.

As for the second issue, the size of a slot is not saved, but could easily be added to the config structure (I've discussed this with someone previously), however does it make sense to? If you just want to avoid another rom slot you can check where they are from the rom table, if you want to avoid overwriting a spiffs or something else on the flash that isn't a rom they aren't tracked by rBoot at all, you reply on the developer knowing something about their flash layout and writing the spiffs in a sensible location, rBoot doesn't know where that will be to put it in the config.

closed 2015-11-29 by hreintke