SmingHub/Sming #364: MacOS start up instructions out of date

ericmnel, 2015-10-22:
I'm following these instructions:

https://github.com/SmingHub/Sming/wiki/MacOS-Quickstart

And when I get to the following command, it references a file that does not exist:

Erics-Mac-Pro:~ admin$ sudo chmod -R 775 /opt/esp-open-sdk
chmod: /opt/esp-open-sdk: No such file or directory
Erics-Mac-Pro:~ admin$ ls -alF /opt
total 8
drwxrwxr-x   7 root        wheel   238 Oct 22 17:10 ./
drwxr-xr-x  41 root        wheel  1462 Oct 22 16:58 ../
drwxr-xr-x  13 ericnelson  staff   442 Jul 11 14:46 esp_iot_sdk_v1.2.0/
drwxr-xr-x   8 ericnelson  staff   272 Jul 11 14:05 esptool/
drwxr-xr-x   3 admin       staff   102 Oct 22 16:58 homebrew-cask/
lrwxr-xr-x   1 ericnelson  staff    18 Jul 11 14:46 sdk@ -> esp_iot_sdk_v1.2.0
drwxr-xr-x   9 ericnelson  staff   306 Jul 11 14:46 xtensa-lx106-elf/

I am assuming that this command (and also the PListBuddy command later in the sequence) should actually be referencing /opt/esp_iot_sdk_v1.2.0 - is that true or did I mess up the install somewhere along the line?

piperpilot, 2015-10-23:
Eric, it looks like your install did not create the sep-open-sdk directory which is standard...all of the stuff in /opt like:

esp_iot_sdk_v1.2.0 and esptool and xtensa-lx106-elf 

are normally inside the esp-open-sdk directory.  A simple fix would be:

cd /opt
sudo mkdir esp-open-sdk
sudo mv esp* esp-open-sdk
sudo mv xten* esp-open-sdk
sudo mv sdk esp-open-sdk

The last one MIGHT not work because it is a symlink...if it doesn't work, let me know here and I'll give you the command to recreate it inside of the esp-open-sdk folder.

kireevco, 2015-10-23:
Sorry to hear Mac QuickStart didn't work. I'll review the procedure and update QuickStart.

ericmnel, 2015-10-23:
Thanks!   That helped a lot.   Also I had to account for the Eclipse application being in /opt/homebrew-cask/Caskroom/eclipse-cpp/4.5.1/Eclipse.app rather than /Applications, so probably add an instruction to move it or reference the brew version.

closed 2015-10-23 by ericmnel