SmingHub/Sming #365: readme and wiki files have absolute links

hreintke, 2015-10-23:
The readme.md and wikipages contain links with absolute addresses.
f.e. https://github.com/anakod/Sming/wiki/MacOS-Quickstart from readme.md

When someone forks the sminghub repository his/her fork will contain wrong links.
By following a link he/she will be redirected to main repo 
We could solve that by using relative links

../../wiki/MacOS-Quickstart

I am not sure how to test in my fork without (in)direct updating/pushing to sminghub/sming repo

kireevco, 2015-10-30:
@hreintke I agree on README.

Wiki pages are using github internal links:
[image]

raburton, 2015-10-31:
When you create a fork you don't get the wiki anyway.

closed 2015-11-04 by kireevco

hreintke, 2015-11-04:
@kireevco : 
Just closing or do you take the action to update README ?

kireevco, 2015-11-04:
as @raburton said fork won't get a wiki. I don't see a reason to update README. Wikipages are relative to wiki (internal links), so they should be fine.

kireevco, 2015-11-04:
Actually one change worth making - change anakod to SmingHub in all links.
I made this change.