SmingHub/Sming #367: Wiring compatibility

tprochazka, 2015-10-24:
I tried to port some lib from Arduino to Sming and I found that Wiring implemented in the Sming is completely different (Or very old?) compared to current esp8266/Arduino implementation

It is because of historical reason? I think that Sming is not so old for this huge difference from Arduino.

There is any plan to make it more compatible?

tprochazka, 2015-11-08:
The same problem with Serial, it is not possible to configure anything except baudrate.

I found the library which using this
```Serial1.begin(baud, SERIAL_6N1, SERIAL_TX_ONLY);```

I really think that will be better copy all basic stuff like this from ESP8266 Arduino project and focus on the benefits of Sming framework.

What others think?

zhivko, 2015-12-09:
Agree.

tprochazka, 2015-12-09:
Yes, for example ScreenOLED_SSD1306 use SPI via Adafruit_SSD1306 library.
For DC,RST and CS you can use any PIN, just MOSI (GPIO13) and CLK (GPIO14) are fixed if you want to use HW SPI.

hreintke, 2016-01-27:
Closed due to low activity

closed 2016-01-27 by hreintke