SmingHub/Sming #368: SPI.setDataMode(SPI_MODE3) is missing

tprochazka, 2015-10-25:
SPI implementation in Sming don't allow to change mode of SPI communication.
I prevent to make working for example OLED display with SSD1331
Arduino implementation of SPI for ESP8266 has this method.
But my first attempt to use this failed. I have no signal on CLK and MOSI pin.

This issue is subtask of https://github.com/SmingHub/Sming/issues/367

mralexgray, 2016-02-16:
`SPI` is missing a LOT of functions...  compare the `sming` implementation...

    void begin(); // Default begin(2, 4)
    void begin(uint16_t predivider, uint8_t divider);
    void end();
    void transfer(uint8_t * data, uint8_t count);
    byte transfer(uint8_t data);

with esp866/arduino's repertoire...

```
  void begin();
  void end();
  void setHwCs(bool use);
  void setBitOrder(uint8_t bitOrder);
  void setDataMode(uint8_t dataMode);
  void setFrequency(uint32_t freq);
  void setClockDivider(uint32_t clockDiv);
  void beginTransaction(SPISettings settings);
  uint8_t transfer(uint8_t data);
  uint16_t transfer16(uint16_t data);
  void write(uint8_t data);
  void write16(uint16_t data);
  void write16(uint16_t data, bool msb);
  void write32(uint32_t data);
  void write32(uint32_t data, bool msb);
  void writeBytes(uint8_t * data, uint32_t size);
  void writePattern(uint8_t * data, uint8_t size, uint32_t repeat);
  void transferBytes(uint8_t * out, uint8_t * in, uint32_t size);
  void endTransaction(void);
```

This feature disparity prevents MOST uses of third party `SPI` libs.  What is the exact nature of the problem with the port?  can we help?

hreintke, 2016-03-25:
Solved by new implementation of generic SPI Classes

closed 2016-03-25 by hreintke