SmingHub/Sming #369: Project targets for examples

hreintke, 2015-10-25:
On esp forum there was a remark on make targets in applications.
Especially on flashinit, flashonefile and flashboot.

I answered : 
```
Flashboot and flashonefile are not valid targets anymore, they should be removed from project or get new functionality.
Flashinit creates an "empty" esp. Used when you have "serious trouble" with flashed applications.
Need to check whether it's valid for all esp's out there.
```
Don't know about the history of these targets. But if we can define the ones we need and correct the working in the makefile(s) I can document and update the project files.

So Question : What application make targets are we supporting ?

raburton, 2015-10-25:
Ditch most, but the ability to erase the last 4 sectors/ wrote blank to them is handy to clear the sdk configuration which can get corrupt and cause problems on occasion. Flash location is dependant on the correct spi size being set.

hreintke, 2015-10-25:
Agree, 
Clear SDK target then should be part of framework makefile instead of project makefile on can we add at one location and have both ?

raburton, 2015-10-25:
Shouldn't be in the framework makefile, no user settings like spi size are set there.

hreintke, 2015-10-26:
But "Resetting" a esp flash is not really a "application specific" action.
Is there a possibility that we include the flashinit target in one of the "global" mk files like makefile-project.mk. makefilewindows.mk ....
Then it would available for all application without adding them to project makefiles.

raburton, 2015-10-26:
Makefile-project.mk / Makefile-rboot.mk is exactly where it should be.

hreintke, 2015-10-27:
Ok, I checked again the implementation and understand now.
I will submit a PR to get the finishboot and flashonefile (not present anymore) out of the eclipse config's.

kireevco, 2015-11-04:
@hreintke any update?

hreintke, 2015-11-04:
No, Other priorities prevailed.

hreintke, 2016-01-27:
closed due to low activity

closed 2016-01-27 by hreintke