SmingHub/Sming #370: Sming and Eclipse Arduino C++ IDE

hreintke, 2015-10-25:
@All,

Recently I read a blog entry of a beta version of the "Arduino C++ IDE for Eclipse" (http://cdtdoug.ca/2015/10/17/introducing-the-arduino-c-ide-for-eclipse/)
Besides my multiple ESP's I also own an Arduino Nano, on which I had my first micro controller experience. In that first experience I was surprised by the "lack of a decent IDE" for development.
This plugin seems to improve that a lot.

Another item that attracted my attention  was the support for ESP programming using the Arduino ESP8266 Project. Meaning Arduino CDT plugin was supporting multiple platforms and even esp8266 toolchain.

That is why I asked Doug Schaefer (creator if the plugin) the following question : 

>– ESP toolkit/framework
For ESP8266 there are multiple frameworks to create applications.
Which of the frameworks is used currently in the plugin ?
Maybe same question on upgrade -> how would you/user handle the upgrade ?

>I am co-developer of the “Sming” framework for ESP8266.
Is there a way that I can integrate/configure (the use of) this framework in the plugin without the need of updating CDT plugin ?

Reply from Doug : 
>– ESP8266

>Right now it’s using the Arduino ESP8266 framework that I get from here: http://arduino.esp8266.com/stable/package_esp8266com_index.json

>If you provide a similar file and related boards.txt and platform.txt files and add it to the URL’s list in the C++/Arduino preference page, it should just work, in theory. The new project wizard does assume you’re doing Arduino development so the initial C++ file is set up that way.

>Interesting thought I’ve had for a while was how flexible the Arduino system is for adding SDK’s toolchains and libraries. It would be interesting if we could generalize this to support other systems like Sming.

>BTW, Sming looks awesome. Great work! I’ll probably take a look at it as I get into programming my ESP8266’s.

My conclusion :  He is positive and looks willing to investigate how sming/other frameworks can be integrated in his CDT plugin.

Remarks :
- It is Eclipse only
- It will take (a lot of ?) effort to get Sming/the plugin included and working correctly.
- It can/will have negative impact on Sming when integration fails due to not supported by Sming Community 

But..
- It can be that I am to positive on the possibilities and willingness of the opportunity.
- It would give Sming tremendous visibility
- It would give Sming an additional distribution platform
- It has multiple hostenvironment support (linux, mac, windows)
- It is supported by an experienced CDT developer/manager and even Eclipse Foundation attention.
- I am willing to put energy in that but it needs **"Full support, participation and commitment from all of you"** to get it started an finished in a proper way.

Question : 
Are we willing to investigate the effort into the collaboration with CDT plugin activities.

Would like to get your ideas/pro's/con's on the above, especially also from @anakod

kireevco, 2015-10-25:
I like the idea. I will look into packaging sming the way arduino ide and this plugin requires. Shouldn't be that hard. 

Main question for me is sdk and took chain for Windows. Still investigating,

alonewolfx2, 2015-10-26:
i like the idea too. i was tried that plugin when its new and i had many problems. i dont know what is status on today but we can integrate it i think.

igrr, 2015-11-01:
Looking at Sming makefiles, it seems quite possible to create a arduino-style platform.txt/boards.txt package after some refactoring. Let me know guys if you need any help with this one.

hreintke, 2015-11-01:
@igrr : 
Would be a good extension of the Sming package, and indeed within the Sming communiy there is not much knowledge on that.
If you can share your ideas on way of implementation and options you see we can decide on a way forward.

kireevco, 2015-11-01:
@igrr, that would be great. I'm working on building some pre-packaged sming, so maybe you could share your thoughts what is the structure requirement to be picked up by arduinoide (and eclipse arduino)? I could provide such packaging and host it as well.

Thank you.

chall3ng3r, 2015-11-07:
Just started playing with Sming, it's awesome, great job @anakod and the rest of contributors!

I like this idea. I've been testing Eclipse Arduino C++ from early builds, and helped fixing some core issues. Doug (the plugin author) is really active on fixing issues, and today the plugin is a lot stable.

I believe Arduino (and community working on projects like, ESP8266 Arduino, Sming, etc.) need to formalize this IDE effort into one painless experience. 

I am willing to beta test any development on this, and provide valuable feedback.

hreintke, 2016-01-27:
Closed due to low activity

closed 2016-01-27 by hreintke