SmingHub/Sming #372: [Basic_rBoot] esptool2: Command not found

fvpalha, 2015-10-26:
Hi.

I have a error on building the rBoot example with UDK 2.0.8:

```
16:29:12 **** Build of configuration Sming for project Sming_Basic_rBoot ****
make all 
make -C /C/Trabalho/GitHub/Sming/Sming/rboot
make[1]: Entering directory `/C/Trabalho/GitHub/Sming/Sming/rboot'
FW /c/Trabalho/GitHub/Sming/Basic_rBoot/out/build/rboot-hex2a.h
make[1]: esptool2: Command not found
make[1]: *** [/c/Trabalho/GitHub/Sming/Basic_rBoot/out/build/rboot-hex2a.h] Error 127
make[1]: Leaving directory `/C/Trabalho/GitHub/Sming/Sming/rboot'
make: *** [out/firmware/rboot.bin] Error 2
```

Kind regards.

fvpalha, 2015-10-26:
Hi,

I copied the "esptool2.exe" to folder Basic_rBoot and I edited the Makefile-user.mk, but:

```
16:53:39 **** Build of configuration Sming for project Sming_Basic_rBoot ****
make all 
OC out/build/libmain2.a
make -C /C/Trabalho/GitHub/Sming/Sming/rboot
make[1]: Entering directory `/C/Trabalho/GitHub/Sming/Sming/rboot'
CC rboot-stage2a.c
LD /c/Trabalho/GitHub/Sming/Basic_rBoot/out/build/rboot-stage2a.elf
FW /c/Trabalho/GitHub/Sming/Basic_rBoot/out/build/rboot-hex2a.h
CC rboot.c
LD /c/Trabalho/GitHub/Sming/Basic_rBoot/out/build/rboot.elf
FW /c/Trabalho/GitHub/Sming/Basic_rBoot/out/firmware/rboot.bin
make[1]: Leaving directory `/C/Trabalho/GitHub/Sming/Sming/rboot'
C+ app/application.cpp
C+ /C/Trabalho/GitHub/Sming/Sming/appinit/user_main.cpp
CC /C/Trabalho/GitHub/Sming/Sming/rboot/appcode/rboot-api.c
In file included from c:/Trabalho/GitHub/Sming/Sming/rboot/appcode/rboot-api.c:17:0:
c:/Trabalho/GitHub/Sming/Sming/rboot/appcode/rboot-api.c: In function 'rboot_set_config':
C:\Espressif\ESP8266_SDK/include/mem.h:8:21: error: implicit declaration of function 'pvPortMalloc' [-Werror=implicit-function-declaration]
 #define os_malloc   pvPortMalloc
                     ^
c:/Trabalho/GitHub/Sming/Sming/rboot/appcode/rboot-api.c:44:19: note: in expansion of macro 'os_malloc'
  buffer = (uint8*)os_malloc(SECTOR_SIZE);
                   ^
C:\Espressif\ESP8266_SDK/include/mem.h:9:21: error: implicit declaration of function 'vPortFree' [-Werror=implicit-function-declaration]
 #define os_free     vPortFree
                     ^
c:/Trabalho/GitHub/Sming/Sming/rboot/appcode/rboot-api.c:63:2: note: in expansion of macro 'os_free'
  os_free(buffer);
  ^
c:/Trabalho/GitHub/Sming/Sming/rboot/appcode/rboot-api.c: In function 'rboot_write_flash':
C:\Espressif\ESP8266_SDK/include/mem.h:10:21: error: implicit declaration of function 'pvPortZalloc' [-Werror=implicit-function-declaration]
 #define os_zalloc   pvPortZalloc
                     ^
c:/Trabalho/GitHub/Sming/Sming/rboot/appcode/rboot-api.c:106:20: note: in expansion of macro 'os_zalloc'
  buffer = (uint8 *)os_zalloc(len + status->extra_count);
                    ^
In file included from c:/Trabalho/GitHub/Sming/Sming/rboot/appcode/rboot-api.c:18:0:
C:\Espressif\ESP8266_SDK/include/osapi.h:16:19: error: implicit declaration of function 'ets_memcpy' [-Werror=implicit-function-declaration]
 #define os_memcpy ets_memcpy
                   ^
c:/Trabalho/GitHub/Sming/Sming/rboot/appcode/rboot-api.c:113:2: note: in expansion of macro 'os_memcpy'
  os_memcpy(buffer, status->extra_bytes, status->extra_count);
  ^
cc1.exe: all warnings being treated as errors
make: *** [out/build//C/Trabalho/GitHub/Sming/Sming/rboot/appcode/rboot-api.o] Error 1

16:54:02 Build Finished (took 22s.917ms)

```

flexiti, 2015-10-26:
It was corrected in Sming develop, What Sming version have You ?

flexiti, 2015-10-26:
or just include:
 #include system/include/esp_systemapi.h

fvpalha, 2015-10-26:
I used the master version.

fvpalha, 2015-10-26:
@flexiti where I put this code?
```
#include system/include/esp_systemapi.h
```

flexiti, 2015-10-26:
rboot-api.c,

or download all new develop sming

fvpalha, 2015-10-26:
Ok. Thank you.

flexiti, 2015-10-26:
#338

closed 2015-10-27 by fvpalha