SmingHub/Sming #373: [Basic_rBoot] spiffy: No such file or directory - Error 127

fvpalha, 2015-10-27:
Hi.

I download all new develop Sming, I build the sming library and I edited the Basic_rboot Makefile-user.mk and user_config.h.

But this error occurs:

```
10:28:12 **** Build of configuration Sming for project Sming_Basic_rBoot-develop ****
make all 
OC out/build/libmain2.a
make -C /C/Trabalho/GitHub/Sming-develop/Sming/rboot
make[1]: Entering directory `/C/Trabalho/GitHub/Sming-develop/Sming/rboot'
CC rboot-stage2a.c
xtensa-lx106-elf-gcc -Os -O3 -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals  -D__ets__ -DICACHE_FLASH -DBOOT_BIG_FLASH  -c rboot-stage2a.c -o /c/Trabalho/GitHub/Sming-develop/Basic_rBoot/out/build/rboot-stage2a.o
LD /c/Trabalho/GitHub/Sming-develop/Basic_rBoot/out/build/rboot-stage2a.elf
FW /c/Trabalho/GitHub/Sming-develop/Basic_rBoot/out/build/rboot-hex2a.h
CC rboot.c
LD /c/Trabalho/GitHub/Sming-develop/Basic_rBoot/out/build/rboot.elf
FW /c/Trabalho/GitHub/Sming-develop/Basic_rBoot/out/firmware/rboot.bin
make[1]: Leaving directory `/C/Trabalho/GitHub/Sming-develop/Sming/rboot'
C+ app/application.cpp
CC /C/Trabalho/GitHub/Sming-develop/Sming/rboot/appcode/rboot-api.c
CC /C/Trabalho/GitHub/Sming-develop/Sming/rboot/appcode/rboot-bigflash.c
AR out/build/app_app.a
LD out/build/app_0.out
E2 out/firmware/rom0.bin
Checking for spiffs files
files directory exists. Creating out/firmware/spiff_rom.bin
/bin/sh: line 2: /C/Trabalho/GitHub/Sming-develop/Sming/spiffy/spiffy: No such file or directory
make: *** [out/firmware/spiff_rom.bin] Error 127

10:28:48 Build Finished (took 36s.28ms)
```

What is wrong?

Kind regards.

raburton, 2015-10-27:
Did you build spiffy? See the release notes: https://github.com/SmingHub/Sming/releases

fvpalha, 2015-10-27:
@raburton Really I don't build spiffy.

But, when I build:

```
C:\Trabalho\GitHub\Sming-develop\Sming>make spiffy
Making spiffy utility
CC spiffy.c
CC ../Services/SpifFS/spiffs_cache.c
CC ../Services/SpifFS/spiffs_nucleus.c
CC ../Services/SpifFS/spiffs_hydrogen.c
../Services/SpifFS/spiffs_hydrogen.c: In function 'SPIFFS_mount':
../Services/SpifFS/spiffs_hydrogen.c:80: error: #pragma GCC diagnostic not allow
ed inside functions
../Services/SpifFS/spiffs_hydrogen.c:81: error: #pragma GCC diagnostic not allow
ed inside functions
../Services/SpifFS/spiffs_hydrogen.c:82: warning: cast from pointer to integer o
f different size
../Services/SpifFS/spiffs_hydrogen.c:83: error: #pragma GCC diagnostic not allow
ed inside functions
../Services/SpifFS/spiffs_hydrogen.c:92: error: #pragma GCC diagnostic not allow
ed inside functions
../Services/SpifFS/spiffs_hydrogen.c:93: error: #pragma GCC diagnostic not allow
ed inside functions
../Services/SpifFS/spiffs_hydrogen.c:94: warning: cast from pointer to integer o
f different size
../Services/SpifFS/spiffs_hydrogen.c:95: error: #pragma GCC diagnostic not allow
ed inside functions
make[1]: *** [spiffs_hydrogen.o] Error 1
make: *** [spiffy/spiffy] Error 2

C:\Trabalho\GitHub\Sming-develop\Sming>
```

raburton, 2015-10-27:
Never seen that before, what version of gcc (for x86/x64, not extensa) do you have?

fvpalha, 2015-10-27:
I have two versions in MinGW (4.4.0 and 4.8.1).

raburton, 2015-10-27:
Use 4.8.1, should work fine with that. 4.4 is too old to support that code.

fvpalha, 2015-10-27:
I will update the MinGW, which version you recomend? (5.2.0) ?

raburton, 2015-10-27:
I recently tested whatever was the latest (2-3 weeks ago), 5.2.0 rings a bell so that should be fine. But your existing 4.8.1 should work.

fvpalha, 2015-10-27:
Thank you @raburton I will try the version 5.2.0, if I have a problem I switch to 4.8.1

fvpalha, 2015-10-27:
Thank you @raburton .

```
C:\Trabalho\GitHub\Sming-develop\Sming>make spiffy
Making spiffy utility
CC ../Services/SpifFS/spiffs_hydrogen.c
CC ../Services/SpifFS/spiffs_gc.c
CC ../Services/SpifFS/spiffs_check.c
LD spiffy
Done

C:\Trabalho\GitHub\Sming-develop\Sming>gcc --version
gcc (GCC) 4.8.1
```

closed 2015-10-27 by fvpalha