SmingHub/Sming #376: .settings junk

raburton, 2015-10-27:
All the samples seem to have a folder called .settings that is full of eclipse stuff, but it doesn't seem to be important project stuff, just user settings. Can Anyone who uses eclipse confirm that these aren't needed and I'll strip them out.

hreintke, 2015-10-27:
I am busy cleaning up and correcting the eclipse settings/options for both framework and examples.
Will submit a PR for that shortly.

raburton, 2015-10-27:
Cool.

raburton, 2015-10-27:
I can pull all those out and add the necessary to the .gitignore for you if you like. I've got a few minutes spare...

kireevco, 2015-10-28:
Thank you, @hreintke I assume you remember about cross-platform needs?

hreintke, 2015-10-28:
@kireevco : 
I make sure that I have the used paths only dependent on the SMING_HOME and ESP_HOME.
That is also how the current implementation works.
Eclipse itself is multiplatform, so should this should work on all.

Do I miss something in this reasoning  ?

kireevco, 2015-10-28:
@hreintke 
> I make sure that I have the used paths only dependent on the SMING_HOME and ESP_HOME.

Yes, and also we need to make sure that path separators work fine on all systems. But they should be ok.

kireevco, 2015-11-04:
@hreintke any update?

hreintke, 2015-11-04:
No, Other priorities prevailed

raburton, 2015-11-08:
done

closed 2015-11-08 by raburton