SmingHub/Sming #377: [Basic_rBoot] switch - Rom 1 is bad.

fvpalha, 2015-10-27:
Hi.

I download all new develop Sming.

I build and flash, but I don't can switch rom:

```
trying to mount spiffs at 0x40300000, length 65536
fs.start: size:64 Kb, offset:0x0x100000
mount res: 0

Currently running rom 0.
Type 'help' and press enter for instructions.
mode : null

SDK: v1.3.0
Free Heap: 45280
CPU Frequency: 80 MHz
System Chip ID: 0x9fd8b1
SPI Flash ID: 0x1640c8

Swapping from rom 0 to rom 1.
Restarting...
 ets Jan  8 2013,rst cause:1, boot mode:(3,3)
load 0x40100000, len 1516, room 16 
tail 12
chksum 0xd0
ho 0 tail 12 room 4
load 0x3ffe8000, len 700, room 12 
tail 0
chksum 0x98
csum 0x98

rBoot v1.2.1 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Rom 1 is bad.

Booting rom 0.

r

trying to mount spiffs at 0x40300000, length 65536
fs.start: size:64 Kb, offset:0x0x100000
mount res: 0

Currently running rom 0.

Type 'help' and press enter for instructions.
mode : null
```

What is wrong?

raburton, 2015-10-27:
Did you flash rom1?

alonewolfx2, 2015-10-28:
@fvpalha you need to flash rom1 after first upload firmware or you need to use ota after first firmware upload for writing rom1. after that you can use rom switch function.

flexiti, 2015-10-28:
>System Chip ID: 0x9fd8b1
>SPI Flash ID: 0x1640c8
>Swapping from rom 0 to rom 1.  <-----

Why You swapping roms here ? You have flashed rom 0 only via serial I suppose,
Please write what you want to do, it will be easier hint 
Have You 2 different roms with 2 different programs ? or You testing ota ?

fvpalha, 2015-10-28:
Thank you @raburton 

The Makefile-rboot.mk flash only rom0:
```
flash: all
	$(vecho) "Killing Terminal to free $(COM_PORT)"
	-$(Q) $(KILL_TERM)
ifeq ($(DISABLE_SPIFFS), 1)
# flashes rboot and first rom
	$(ESPTOOL) -p $(COM_PORT) -b $(COM_SPEED_ESPTOOL) write_flash $(flashimageoptions) 0x00000 $(RBOOT_BIN) 0x02000 $(RBOOT_ROM_0)
else
# flashes rboot, first rom and spiffs
	$(ESPTOOL) -p $(COM_PORT) -b $(COM_SPEED_ESPTOOL) write_flash $(flashimageoptions) 0x00000 $(RBOOT_BIN) 0x02000 $(RBOOT_ROM_0) $(RBOOT_SPIFFS_0) $(SPIFF_BIN_OUT)
endif
	$(TERMINAL)
```

raburton, 2015-10-28:
> The Makefile-rboot.mk flash only rom0

Yes, that's correct. If you want the second rom installed do an ota or flash it manually.

closed 2015-10-28 by raburton

fvpalha, 2015-10-28:
Thank you.

I followed the tips, now all works.

```
TCP received: 1368 bytes
onReadyToSendData: 1
TCP received: 197 bytes
onReadyToSendData: 1
TCP received: (null)
TCP connection closing
-TCP connection
Download file:
    (1) http://192.168.0.42:8080/spiff_rom.bin -> 0x100000
Download: http://192.168.0.42:8080/spiff_rom.bin
+TCP connection
connect to: 192.168.0.42
TcpConnection::connect result:, 0
realloc 49 -> 180
OnConnected
TCP connected
onReadyToSendData: 0
TcpClient request completed
TCP sent: 51
onReadyToSendData: 2
Header pos: 201
Date === Wed, 28 Oct 2015 10:39:45 GMT
Server === Apache/2.4.10 (Win32) OpenSSL/1.0.1h PHP/5.4.31
Content-Length === 314
Connection === close
Content-Type === text/html; charset=iso-8859-1
TCP received: 519 bytes
onReadyToSendData: 1
TCP received: (null)
TCP connection closing
-TCP connection
Firmware download failed..
In callback...
Firmware update failed!

Swapping from rom 1 to rom 0.
Restarting...

state: 5 -> 0 (0)
rm 0
pm close 7 0 0/40424325
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:1, boot mode:(3,3)
load 0x40100000, len 1516, room 16 
tail 12
chksum 0xd0
ho 0 tail 12 room 4
load 0x3ffe8000, len 700, room 12 
tail 0
chksum 0x98
csum 0x98

rBoot v1.2.1 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Booting rom 0.
r
trying to mount spiffs at 0x40300000, length 65536
fs.start: size:64 Kb, offset:0x0x100000
mount res: 0

Currently running rom 0.
Type 'help' and press enter for instructions.
mode : sta(18:fe:34:9f:d8:b1)
add if0
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
pm open phy_2,type:2 0 0
cnt 

connected with dd-wrt, channel 6
dhcp client start...
ip:192.168.1.126,mask:255.255.255.0,gw:192.168.1.1

Currently running rom 0.

Swapping from rom 0 to rom 1.
Restarting...
state: 5 -> 0 (0)
rm 0
pm close 7 0 0/14361726
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:1, boot mode:(3,3)
load 0x40100000, len 1516, room 16 
tail 12
chksum 0xd0
ho 0 tail 12 room 4
load 0x3ffe8000, len 700, room 12 
tail 0
chksum 0x98
csum 0x98

rBoot v1.2.1 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Booting rom 1.
r

trying to mount spiffs at 0x40500000, length 65536
fs.start: size:64 Kb, offset:0x0x300000
mount res: 0

Currently running rom 1.
Type 'help' and press enter for instructions.
mode : sta(18:fe:34:9f:d8:b1)
add if0
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
pm open phy_2,type:2 0 0
cnt 

connected with dd-wrt, channel 6
dhcp client start...
ip:192.168.1.126,mask:255.255.255.0,gw:192.168.1.1

Currently running rom 1.
```

fvpalha, 2015-10-28:
I forgot the "spiff_rom.bin" file. 

I added the file in root folder of webserver.

```
TCP received: 1368 bytes
onReadyToSendData: 1
TCP received: 164 bytes
onReadyToSendData: 1
TCP received: (null)
TCP connection closing
-TCP connection

Firmware download finished!
 - item: 0, addr: 0x2000, len: 240672 bytes
 - item: 1, addr: 0x100000, len: 65536 bytes
Firmware updated.
In callback...

Firmware updated, rebooting to rom 0...

state: 5 -> 0 (0)
rm 0
pm close 7 0 0/635054902
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:1, boot mode:(3,3)
load 0x40100000, len 1516, room 16 
tail 12
chksum 0xd0
ho 0 tail 12 room 4
load 0x3ffe8000, len 700, room 12 
tail 0
chksum 0x98
csum 0x98

rBoot v1.2.1 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Booting rom 0.
r

trying to mount spiffs at 0x40300000, length 65536
fs.start: size:64 Kb, offset:0x0x100000

First init file system
sect_first: 0x100, sect_last: 0x110
0%
6%
12%
18%
25%
31%
37%
43%
50%
56%
62%
68%
75%
81%
87%
93%
100%
formated
mount res: 0

Currently running rom 0.
Type 'help' and press enter for instructions.
mode : sta(18:fe:34:9f:d8:b1)
add if0
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
pm open phy_2,type:2 0 0
cnt 

connected with dd-wrt, channel 6
dhcp client start...
ip:192.168.1.126,mask:255.255.255.0,gw:192.168.1.1
```

samschaerer, 2016-05-01:
Hi
How I can switch between 2 roms without OTA. Where I have to flash rom1 and rom2 manually (address)?

Thanks

hreintke, 2016-05-01:
@samschaerer : 
Better to open a new issue as commenting on an old, closed one.

Flash location is dependent on your esp spi_size.
But if you have a 4MB esp with default OTA settings the application locations are 0x000200 and 0x202000.

samschaerer, 2016-05-02:
@hreintke :
Thanks for the hint and your answer