SmingHub/Sming #378: Possible memory leak in JsonObject.cpp

linlun, 2015-10-27:
I have a program that is returning an JSON object as a response to a httprequest looking like this:
```cpp
void onApiSensors(HttpRequest &request, HttpResponse &response)
{
	JsonObjectStream* stream = new JsonObjectStream();
	JsonObject& json = stream->getRoot();
	json["status"] = (bool)true;
	json["heap"] = system_get_free_heap_size();
	sensors["temperature"] = tempSensor.get();
	for( int i = 0; i < tempSensor.numSensors; i++)
	{
		char buff[3];
		itoa(i, buff, 10);
		String desiredString = "sensor_";
		desiredString += buff;
		sensors[desiredString] = tempSensor.get(i);
	}
	JsonObject& pid = json.createNestedObject("pid");
	pid["Setpoint"] = AppSettings.Setpoint;
	response.sendJsonObject(stream);
}
```
But on each request the heap size get smaller, the issue has been tracked down to the line `sensors[desiredString] = tempSensor.get(i);`. 
It seems as the Json implementation does not free dynamic memory as it should, the problem is most likely in https://github.com/SmingHub/Sming/blob/master/Sming/Services/ArduinoJson/src/JsonObject.cpp on row 44 where the memory is not freed.

A new version looking like this does not have an issue with memory leakage:
```cpp
const char* names[] = { "sensor_01", "sensor_02", "sensor_03", "sensor_04", "sensor_05", "sensor_06" };
void onApiSensors(HttpRequest &request, HttpResponse &response)
{
	JsonObjectStream* stream = new JsonObjectStream();
	JsonObject& json = stream->getRoot();
	json["status"] = (bool)true;
	json["heap"] = system_get_free_heap_size();
	JsonObject& sensors = json.createNestedObject("sensors");
	sensors["temperature"] = tempSensor.get();
	for( int i = 0; i < tempSensor.numSensors; i++)
	{
		sensors[names[i]] = tempSensor.get(i);
	}
	JsonObject& pid = json.createNestedObject("pid");
	pid["Setpoint"] = AppSettings.Setpoint;
	response.sendJsonObject(stream);
}
``` 

Regards Linus

raburton, 2015-10-28:
This looks to be code that was written by someone else and pulled into this project. We should really be looking to update our code to the newest upstream version where the bug may already be fixed. Have you had a look upstream to see if it is fixed or how easily we could move to a newer version?

hreintke, 2015-10-28:
@raburton : 
I did, and that resulted in the conclusions below.
1/ There are multiple new versions of upstream.

We are on 4.2, R4 is up to 4.6.1 -> fairly but not complete compatible
There is also 5.0.6 -> not compatible needs rework

2/ The issue is in code that is added in Sming and not in upstream

3/ I am not sure (yet) if and where other "Sming Additions" have been done.

For the intermediate solution : 
@linlun : 
Can you check whether the construct : `sensors[desiredString.c_str()] = tempSensor.get(i);`
Has correct or incorrect results.

mattiasrunge, 2015-10-28:
@raburton 
Me and @linlun tried that yesterday and it does not work. The result is that we send in a pointer to the library to our scoped String variable (desiredString) which is then stored in the JSON object. But that is then freed when we go out of scope of desiredString and the pointer is not valid anymore and we get corruptions. So that is not a workaround unfortunately.

hreintke, 2015-10-28:
Update on the above : 
Sming-Arduinojason version differs on multiple locations from Standard-ArduinoJson.
These differences are 
- at least for including support for "Sming-String", "Sming-Print" and "Sming-Datasource memorystream" 
- Scattered around multiple ArduinoJason framework files
- Not available as "Patch" to the original (were already included in initial insert in Sming)

I've seen earlier issues with at() function which I now know most certainly will be related also the "Sming-Arduino" version".

@linlun @mattiasrunge : 
The "out-of-scope" issue is what this part of Sming patch is trying to resolve in the overloaded version of operator[] but using some "arduinojason internal fucntions" Now resulting in a memory leak.
Is your knowledge/experience in Arduinojson such that with joined effort we can try to solve this issue in current version for intermediate solution ?

We should reconsider the "Sming-updates" and definitely find a better way to identify these to be able to follow upstream versions.

hreintke, 2015-10-28:
Next update : 
The issue is conceptual in the adding of "Sming-String" option in Arduinojson.
Due to the nature (implementation) of "Sming-String" and in order to prevent the "out-of-scope" issue using them, at lots of locations within the Sming adaptions, a Sming added JsonStringStorage object is created. A pointer to data within this object is the passed/used within json processing. 
This JsonStringStorage object is never destructed/removed.
Do not (yet?) see an more or less easy (intermediate) solution for this.

I am not sure whether we should put effort in solving this or putting effort into integrating the new Release 5 and making that a manageable

linlun, 2015-10-28:
I'm sorry but at least I don't have the experience/knowledge to dig into arduinojson to find a good solution.
For me it sounds as the best solution to integrate the latest version and make sure to document the needed changes, but that is probably alto of work.
Is it possible to remove the possibility to use Sming-string as an option?

hreintke, 2015-10-29:
@linlun @mattiasrunge @piperpilot @robotiko : 
I looked into the ArduinoJson Release 5 and guess what : R5 includes support for the String implementation of Wiring package. That is the same as used within Sming.

I took the latest ArduinioJson R5 from Github and did a first try to include this version into Sming.
I left/updated some of the integration code within SmingCore.
I left the ArduinoJson R5 version unchanged -> should be easier upgrade to newer versions
And added some "Json String usage" into the Http_ajax example
Status : 
- Alpha version, no guarantee final version will have same functionality, working in my environment
- Sming framework compiles 
- The updated HTTP example compiles, runs an produces json responses.
- I did not test any other functionality and/or memory leakage
- It is available in my Sming fork, https://github.com/hreintke/Sming/tree/ArduinoJson_R5_Alpha 

Can you test in your environment and with more use cases ?
Please feedback in this issue and not within my sming fork.
If the implementation looks OK, I will cleanup and structure the integration and then move it to a branch on Sminghub/sming

linlun, 2015-10-29:
I did not have much time to look into this to night, but I failed to compile my program with your branch of Sming, the http-example compiles fine, but as I have also alot of functionality from the HttpServer_ConfigNetwork example alot och the JSON functionality was missing (or has changed syntax)
I will look more into this in the weekend.

hreintke, 2015-10-29:
Thanks for the feedback. I see also the "trouble" with httpserver_configNetwork.
Will check that.

hreintke, 2015-10-30:
Added a patch to my fork, branch Arduinojson_R5_Alpha

The issues in Http_Config_Network are related to the adaptions in Sming to make Stingconversion work
I updated the example to work with "standard ArduinoJson R5"
The Http_Config_Network example is now working in my environment.

Again : 
- First implementation, changes/updates still possible
- Please test with this and your own application(s)
- Feedback in this issue

Main API changes in examples below. 

From :
`item.add("id", (int)networks[i].getHashId());`
To : 
`item["id"] = (int)networks[i].getHashId();`

From : 
```
item.addCopy("title", networks[i].ssid);
item.add("signal", networks[i].rssi);
item.add("encryption", networks[i].getAuthorizationMethodName());
```
To : 
```
item["title"] = networks[i].ssid;
item["signal"] = networks[i].rssi;
item["encryption"] = networks[i].getAuthorizationMethodName();
```
From :
```			
network.addCopy("ip", ip.toString());
network.addCopy("netmask", netmask.toString());
network.addCopy("gateway", gateway.toString());
```
To :
```
network["ip"] = ip.toString();
network["netmask"] = netmask.toString();
network["gateway"] = gateway.toString();
```
From :
```
fileSetContent(APP_SETTINGS_FILE, root.toJsonString());
```
To :
```
String rootString;
root.printTo(rootString);
fileSetContent(APP_SETTINGS_FILE, rootString);
```

vsirvent, 2015-10-30:
Isn't this the same issue as https://github.com/SmingHub/Sming/issues/339?

hreintke, 2015-10-30:
@vsirvent : 
Yes it is. And I trying to do what I suggested : upgrading to the new arduinojson release.
Would be great if you can help testing and give feedback

vsirvent, 2015-10-30:
Hi,

Is there any reason to have two different issues of the same bug?

I just took a look to the new json lib. There are no more dynamic memory allocations inside the "JsonBufferAllocated" classes, so the memory leak should be fixed. However, this is done by using directly references to the program strings instead of copying the value. This is dangerous, as the developer must be aware that the reference must be valid during the life of the json object.

In the example of first comment, the strings are global, so there is no such problem, however something like:
```
JsonObjectStream* stream = new JsonObjectStream();
JsonObject& root = stream->getRoot();

while (bla) {
    String k = bla + bla + bla...
    String v = bla + bla + bla...
    root[k] = v;
}
```

This can crash if i'm not wrong.

I solved the bug in the current json library by tracking the objects created in JsonBuffer into a vector  and calling theirs destructor in the JsonBuffer destructor.

JsonBuffer.h

```
class JsonBuffer {
private:
	Vector<Internals::JsonBufferAllocated*> allocs;
...
```

JsonBuffer.cpp

```

JsonArray &JsonBuffer::createArray()
{
	JsonArray *ptr = new (this) JsonArray(this);
	if (ptr)
	{
		allocs.add(ptr);
	}
	return ptr ? *ptr : JsonArray::invalid();
}

JsonObject &JsonBuffer::createObject()
{
	JsonObject *ptr = new (this) JsonObject(this);
	if (ptr)
	{
		allocs.add(ptr);
	}
	return ptr ? *ptr : JsonObject::invalid();
}

JsonStringStorage& ArduinoJson::JsonBuffer::createStringStorage(
		const String& text)
{
	JsonStringStorage *ptr = new (this) JsonStringStorage(text);
	if (ptr)
	{
		allocs.add(ptr);
	}
	return ptr ? *ptr : JsonStringStorage::invalid();
}

JsonBuffer::~JsonBuffer()
{
	for (int i = 0; i < allocs.size(); ++i)
	{
		delete allocs[i];
	}
}

```

Also added empty destructor to JsonBufferAllocated:

```
class JsonBufferAllocated {
 public:
  virtual ~JsonBufferAllocated(){}
  void *operator new(size_t n, JsonBuffer *jsonBuffer) throw() {
    return jsonBuffer->alloc(n);
  }

  void operator delete(void *, JsonBuffer *) throw() {}

 -->  void operator delete(void *) throw() {}
};
```

linlun, 2015-10-31:
I have now tested my application with the branch hreintke did and it seems to work great, no memory leaks.

hreintke, 2015-10-31:
@vsirvent :
No there is no specific reason why we have two issues for the same bug.

As I mentioned above, the String support is now included in the ArduinoJson Release 5 and at the moment the alha version of integration uses an unmodified ArduinoJson codebase.
That means that we also can submit an issue at the project in case of problems with the implementation.
The documentation within the project addresses memory usage and how String can be used.
http://blog.benoitblanchon.fr/arduino-json-v5-0/

Please test the code you think "crash if i'm not wrong" with the Sming branch.
If indeed I will contact the ArduinoJson project for solving.

If integration of ArduinoJson R5 still fails foe whatever reason we have to look to alternatives, for which your solution might be adequate.

PS : Would have been nice if you had shared your solution before.

hreintke, 2015-10-31:
@linlun : 
Thanks for the feedback. I'll wait for some more feedback and cleanup integration.
When done I will submit the PR to Sminghub/sming main repository for broader testing and merging into development branch for usage in one of the coming Releases.

linlun, 2015-10-31:
I tested this code:
```
void onApiSensors(HttpRequest &request, HttpResponse &response)
{
	JsonObjectStream* stream = new JsonObjectStream();
	JsonObject& json = stream->getRoot();
	json["status"] = (bool)true;
	json["heap"] = system_get_free_heap_size();
	JsonObject& sensors = json.createNestedObject("sensors");
	sensors["temperature"] = tempSensor.get();
	int i = 4;
	while (i--) {
	    String k = "testing" + String(i) +String("_bla");
	    String v = "value"+ String(i+5) +String("_val");
		json[k] = v;
	}
```

And it worked as expected, not sure if i missed something in the example from @vsirvent.

vsirvent, 2015-10-31:
Hi,
@hreintke you are totally right, I did not see this post before  and I thought that people was not working on the bug yet.

@linlun about the sample, I've been trying to use the Sming-ArduinoJson_R5_Alpha but esptool.py fails:

'''
AR out/build/app_app.a
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------
/bin/sh: 1: [[: not found

out/build/app.out:     file format elf32-xtensa-le

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000358  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       00001ebc  3ffe8358  3ffe8358  00000438  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00007500  3ffea218  3ffea218  000022f8  2**4
                  ALLOC
  3 .irom0.text   00038916  40209000  40209000  000079e0  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .text         000056df  40100000  40100000  000022f4  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
------------------------------------------------------------------------------
# Generating image...
Traceback (most recent call last):
  File "/opt/esp-open-sdk/esptool/esptool.py", line 647, in <module>
    image.add_segment(e.get_symbol_addr(start), data)
  File "/opt/esp-open-sdk/esptool/esptool.py", line 384, in get_symbol_addr
    self._fetch_symbols()
  File "/opt/esp-open-sdk/esptool/esptool.py", line 381, in _fetch_symbols
    self.symbols[fields[2]] = int(fields[0], 16)
ValueError: invalid literal for int() with base 16: 'U'
make: *** [out/build/app.out] Error 1

'''
Don't know how to fix this...master Sming works correctly.

Anyway, just check the code when json[k] = v; is called, this calls

inline JsonObjectSubscript<const String &> JsonObject::operator[](
    const String &key) {
  return JsonObjectSubscript<const String &>(*this, key);
}

Which stores the reference of the String "k" in 

JsonObjectSubscript::TKey _key; with TKey ==> const String &

So as long as the string in not deleted in the main app, the key will be valid. What can also happen during an execution is that heap memory storing the strings is freed but the memory is not overwritten, so the application can seem to run correctly, but  you are accessing invalid memory addresses, that can explain why your code is working.

hreintke, 2015-10-31:
@vsirvent : 
Not trying to interpret code and to be sure about the issue I raised the following question at the ArduinoJson project : 
```
In the construction :

{
    StaticJsonBuffer<200> jsonBuffer;

    JsonObject& json = jsonBuffer.createObject();
    int i = 4;
    while (i--) {
        String k = "testing" + String(i) +String("_key");
        String v = "value"+ String(i+5) +String("_val");
        json[k] = v;
    }
    json.printTo(Serial);
}

Which uses local instead of global variables for key and value.

Is this a valid application ?
In other words : Will both Key and Value be copied into the jsonBuffer 
and be kept after Stings k and v are out of scope ?
```
The answer is : 
```
Yes, that's exactly what will happen.

The key is duplicated in JsonObject.ipp, line 194.
The value is duplicated in JsonObject.ipp, line 204.

As a consequence, you probably need to increase the size of the StaticJsonBuffer.
```
I think this solves your worrying about accessing invalid memory addresses

robotiko, 2015-10-31:
Hi @hreintke 
I'm late at the party, but I do fully support the ArduinoJson update rather than patch older version.
There are very serious issues that have been fixed in  5.0.2 >= version but also being up to date and compatible with ArduinoJson  is a great added value because it is very active in maintenance.

vsirvent, 2015-11-01:
@hreintke Thanks for the reply, I'm trying to compile the code even copying manually the new json lib to the master branch but I get the same error with esptool.py, any idea to fix the error?

I could compile it to run on linux directly, so I did some tests. Using the String class is ok, as they told, the String is copied using jsonbuffer->strdup, so it's fine, however if using char* instead of String, they don't use strdup in this case the copy the reference.
```
jsonobject.ipp:189: node->content.key = key;
```
So this code:

```
while (i--) {
		char k[128];
		char v[128];
		sprintf(k,  "key %d", i);
		sprintf(v,  "val %d", i);
		json[k] = v;
		sprintf(k,  "key2 %d", i);
		sprintf(v,  "val2 %d", i);
		str.clear();
		json.prettyPrintTo(str);
		printf("%d ==> %s\n", i, str.c_str());
	}
```
prints this output:

```
3 ==> {
  "key2 3": "val2 3"
}
2 ==> {
  "key2 2": "val2 2"
}
1 ==> {
  "key2 1": "val2 1"
}
0 ==> {
  "key2 0": "val2 0"
}

```

And if you add code to use json when the stack address stores other variables, it overlaps the json key:

```

while (i--) {
		char k[128];
		char v[128];
		sprintf(k,  "key %d", i);
		sprintf(v,  "val %d", i);
		json[k] = v;
		sprintf(k,  "key2 %d", i);
		sprintf(v,  "val2 %d", i);
		str.clear();
		json.prettyPrintTo(str);
		printf("%d ==> %s\n", i, str.c_str());
	}
	{
		unsigned char arr1[32];
		unsigned char arr2[32];
		memset(arr1, 'p', 32);
		memset(arr2, 't', 32);
		str.clear();
		json.prettyPrintTo(str);
		printf("==> %s\n", str.c_str());
	}

```

this outputs:

```
3 ==> {
  "key2 3": "val2 3"
}
2 ==> {
  "key2 2": "val2 2"
}
1 ==> {
  "key2 1": "val2 1"
}
0 ==> {
  "key2 0": "val2 0"
}
==> {
  "tttttttttttttttttttttttttttttttt`\bq��5Ϳ�/s��:s��6Ϳ�6Ϳپq�p6Ϳ��\bX6Ϳt:s�": "pppppppppppppppppppppppppppppppp@�\b��\b"
}
```

With valgrind showing memory errors.

hreintke, 2015-11-01:
@vsirvent : 
Please check documentation of ArduioJson on : 
- http://blog.benoitblanchon.fr/arduino-json-v5-0/
- https://github.com/bblanchon/ArduinoJson

I think what you describe is expected behavior of the package.
My goal is to use the package without local updates.

vsirvent, 2015-11-01:
@hreintke Yes that's right, it's expected not to copy the array then. May be that's the a design discussion, like why String is duplicated and char* is not, It can be painful to developers... 

Anyway, String seems to work great after some tests and solves the dynamic memory allocation that we have in the previous version.

hreintke, 2015-11-01:
@All 
In the Http_Confignetwork - AppSettings. h file there is currently the code : 
```
			JsonObject& network = root["network"];
			ssid = String((const char*)network["ssid"]);
			password = String((const char*)network["password"]);
```
Where I had hoped to have this : 
```
			JsonObject& network = root["network"];
			ssid = String(network["ssid"]);
			password = String(network["password"]);
```
But using this I get the error ar compile.
```
include/AppSettings.h:38:33: error: call of overloaded 'String(ArduinoJson::JsonObjectSubscript<const char*>)' is ambiguous
    ssid = String(network["ssid"]);
```
I have limited knowledge of scope/namespace etc.
Who can help getting this in place ?
Or is it just a fact of life ?

hreintke, 2015-11-01:
@vsirvent : 
I agree with the possibility of  "painful for developers" but I think that that design decision should be held in the ArduinoJson project and Sming will be a "follower"

robotiko, 2015-11-03:
Hi, 
tested the new code and after some minor and pre-known changes  (documented in the release version log https://github.com/bblanchon/ArduinoJson/releases), it seems to be working without issues. 
Code update was simple and fast. 
I  support to move this in core. 
The current version in Sming is very old and has some serious issues.

hreintke, 2015-11-04:
@All 
Thanks for the initial evaluation and the feedback.
I submitted PR #402 for integrating the functionality into Develop Branch.

Please verify whether the update in the PR is also working as expected.

From  on, feedback issues/remarks in the PR.

hreintke, 2015-11-12:
Solved by #402

closed 2015-11-12 by hreintke