SmingHub/Sming #379: Can't link basic blink

ericmnel, 2015-10-29:
Obviously I messed up the install somehow:

18:20:25 **** Incremental Build of configuration Sming for project Basic_Blink ****
make all 
LD out/build/app.out
/opt/esp-open-sdk/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot find -lsming
collect2: error: ld returned 1 exit status
make: *** [out/build/app.out] Error 1

18:20:25 Build Finished (took 91ms)

Any pointers on how I can get esp-open-sdk to notice the Sming library appreciated.   The only way I knowingly deviated from the process was to put Spiffy in /usr/local/bin instead of /usr/bin since it was also in my path and my admin user couldn't write to /usr/bin.

ericmnel, 2015-10-29:
One other deviation - I didn't install Eclipse the way suggested in the notes, but rather via the Eclipse website (the C++ version) since that seemed to put Eclipse under /Applications instead of caskroom.

raburton, 2015-10-29:
Did you build libsming? This is no longer supplied pre-compiled. Spiffy is also now included in Sming, but also needs to be compiled, and should not need to be put anywhere other than where it is compiled.

ericmnel, 2015-10-29:
Thanks!   That got me a lot further.

09:10:08 **** Incremental Build of configuration Sming for project Basic_Blink ****
make all 
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------

out/build/app.out:     file format elf32-xtensa-le

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000541  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       0000065c  3ffe8550  3ffe8550  00000630  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00009378  3ffe8bb0  3ffe8bb0  00000c90  2**4
                  ALLOC
  3 .irom0.text   0002aaa4  40209000  40209000  00006000  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .text         00005367  40100000  40100000  00000c8c  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
------------------------------------------------------------------------------
# Generating image...
Traceback (most recent call last):
  File "/opt/esp-open-sdk/esptool/esptool.py", line 22, in <module>
    import serial
ImportError: No module named serial
make: *** [out/build/app.out] Error 1

09:10:09 Build Finished (took 1s.872ms)

Is there another thing I need to recompile to get the serial module?

ericmnel, 2015-10-29:
Also, does this mean that the separate version of Spiffy that is supposed to go in /usr/bin (that I put in /usr/local/bin) can be deleted?

ericmnel, 2015-10-29:
At some point I'd like to make some major changes to the Mac startup README, but I'll wait until I have more experience with the toolset.

raburton, 2015-10-29:
> Also, does this mean that the separate version of Spiffy that is supposed to go in /usr/bin (that I put in /usr/local/bin) can be deleted?

Yes, you shouldn't use a separate version any more as it might not be compatible. The version that is included with sming is built from the same code as the spiffs support in libsming so it should always be the right version. The makefiles now, by default, will look for the sming version of spiffy in its build location. You can build it with the eclipse target or by running `make spiffy` in the sming/sming dir.

kireevco, 2015-10-29:
What do you think is missing? I'll make changes.

ericmnel, 2015-10-29:
1.  Delete section about Xcode having to be 6.2, it worked fine with 7.1
2. Change the Eclipse section to have people install with the official installer and put it in /Applications, the place that brew puts Eclipse does not match up with the stuff in the "configure Environment variables" section
3. The "install prebuilt ESP Open SDK" section did't work for me without additional commands

cd /opt
sudo mkdir esp-open-sdk
sudo mv esp* esp-open-sdk
sudo mv xten* esp-open-sdk
sudo mv sdk esp-open-sdk

4. The "Get Sming core" needs to be something like:

cd <any working dir>
sudo git clone https://github.com/anakod/Sming.git
cd Sming/Sming
sudo make
sudo make spiffy
cd ../..
sudo cp Sming /opt

5. Add a warning to the "Configure Environment Variables" section that if they do it from a non-administrator account it will silently fail to add the entries to the file.

6.  Remove the install Spiffy section

7.  Change the "Get Sming Examples" section - they can import the git repo directly from Eclipse without using "git clone" from a terminal.

8.  Whatever the fix is for my "import serial" linker error above turns out to be.

Thanks!   Eric

raburton, 2015-10-29:
shouldn't need to run make as root

ericmnel, 2015-10-29:
You don't need to be root, just logged in as an administrator rather than a regular user.  Administrators have fewer privileges, it is a click box in the user creation system preference panel.

I think my linker problem may be due to an old python install, I'm going to follow this advice:

https://wolfpaulus.com/journal/mac/installing_python_osx/

Some other guy fixed the linker problem by updating python, I'll try that - it looks like a pain because Apple has marked python as "restricted" and you need to boot into recovery mode to change it even as root.

kireevco, 2015-10-29:
@ericmnel normally you would use homebrew to install python that differs from your system one.

ericmnel, 2015-10-29:
Is that somewhere in the "build it yourself" instructions for ESP Open SDK?  I used the pre-built install.
Probably good to add that to the README also.

kireevco, 2015-10-29:
@ericmnel this is normal practice to use [homebrew](http://brew.sh/). If you feel that your python is messed up, try doing `brew install python`.

closed 2015-10-29 by ericmnel

Picatek, 2015-11-02:
I have the same original problem see below:
However I did build libsming as sugested :

This is from the build of the SmingFramework
........ 
AR out/build/app_app.a
Installing libsming
Done

It is a fresh, error free sming installation from today(2015-11-02)
What  am I missing?

The error building Basic_Blink:

13:06:46 Build Finished (took 31s.240ms) 
13:12:25 **** Build of configuration Sming for project Basic_Blink ****
make all 
LD out/build/app.out
c:/espressif/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/5.1.0/../../../../xtensa-lx106-elf/bin/ld.exe: cannot find -lsming
collect2.exe: error: ld returned 1 exit status
make: *** [out/build/app.out] Error 1

hreintke, 2015-11-02:
@Picatek : 
Did you set your environment variables ESP_HOME and SMING_HOME ?

icodk, 2015-11-30:
Thanks for your answer and sorry for the delay. I went on vacation.
Yes:
echo %ESP_HOME%
c:\Espressif
echo %SMING_HOME%
C:\tools\sming\Sming