SmingHub/Sming #380: Mixing Basic_rBoot, Basic_Serial and Telnet_TCPServer_TCPClient

fvpalha, 2015-10-30:
Hi guys.

Currently my project have the [ESP8266-transparent-bridge](https://github.com/beckdac/ESP8266-transparent-bridge) code as base.

I want develop an alternative TCP-Serial bridge with Sming, using the OTA feature of Basic_rBoot example.

I don't can use the Serial port to debug.

Someone have any tips to help me?

alonewolfx2, 2015-10-30:
you can use serial for debug. wat is your problem ?

kireevco, 2015-10-30:
This is something I'm looking for as well, if I'm understanding it correctly. 

I've seen 2 so far 
http://tuanpm.net/esp8266-opensouce-firmware-bridge-for-arduino-mcu/

hreintke, 2015-10-30:
@fvpalha : 
Do you want to say "technically I can use serial to debug but I can't because of the nature of my program" ?

If so, you can use the debug implementation shown in Telnet_TCPServer_TCPClient.

```
TelnetServer tnet;
tnet.setDebug(true);
tnet.listen(23);
```
debug then with : 
`Debug.printf(" ....");`
If Telnet server is connected, it will go to telnet.
 
```

hreintke, 2015-10-30:
BTW : What do you mean with "Using the OTA feature of basic_rboot" ?

fvpalha, 2015-10-30:
Thank you @hreintke .

That's exactly this. Debug with serial port do not can be used.

The OTA feature of Basic_rBoot example I need adding to my project.

I will try your tip, debug over telnet.

fvpalha, 2015-10-30:
Thank you @kireevco but I want to use the Sming library.
I think a bridge TCP-UART example will be very usefull.

fvpalha, 2015-10-30:
Thank you @alonewolfx2 I will go join on gitter.

fvpalha, 2015-10-30:
My initial development based on Basic_rBoot example.

The "application.cpp" file:
```
#include <user_config.h>
#include <SmingCore/SmingCore.h>
#include <SmingCore/Network/TelnetServer.h>

// If you want, you can define WiFi settings globally in Eclipse Environment Variables
#ifndef WIFI_SSID
	#define WIFI_SSID "PleaseEnterSSID" // Put you SSID and Password here
	#define WIFI_PWD "PleaseEnterPass"
#endif

void init() {
	TelnetServer tnet;
	tnet.setDebug(true);
	tnet.listen(23);
}
```

That is the output on Serial port:
```
ÿrÿlœ˜rŸÿŒcân€à Œ ‚ìpŒ|Ž‚ßÿìx’ŸÇ’œälp  ònnäÄ;òn„’œä Ž bbl`$`üpònàƒÜÿ à‚‡Àl€ € € bÀnâãn€$’lœ b€Ä>~ònîÄÁŒŽÿl`üœ#‚n€$Œl`ä8`rn|’’nÀŽà‚ÜÀl`9~ÄÀ œ#‚n€$Œ ŒònnîÄÈl`nnàÃÜÿ àr‡âÜ € p p’œ<Äß € bÀÄ¾~ònîÄÁŒÿl`üœ#‚n€rnrŽ’Ÿ;„À â rrßÛ’nÀà rrŒbØ?ònä„ÜÇ<Œ|lÿlb€Ä>brœ ÜÄ‚œbÜÛbŸØònßØ’nànÜâbp~’nc`‚nà{ß0ÄÇÉvì’œä ÿ ŽŒÿllÃnä’`Ä8œÇn€$ln„’Üä ÿ ‚ÛÄùlc`‚nà{ß0ÄÇ x’Üb ÿ Ž€ rÄ1’`Ä> Ûnnä„? xnnßb  ›äùN€ŒœxÜÇnc`cà ›nnänßòœÿŒ|~’nÀ|l`ür+TCP connection
timeout updating: 70 -> 65535
DebugClass Instantiating
+TCP connection
timeout updating: 70 -> 65535
TCP connection closing
~TCP connection
mode : sta(18:fe:34:9f:d8:b1)
add if0
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
pm open phy_2,type:2 0 0
cnt 

connected with PleaseEnterSSID, channel 6
dhcp client start...
ip:192.168.1.126,mask:255.255.255.0,gw:192.168.1.1
```

hreintke, 2015-10-30:
@fvpalha : 
If you have not yet experienced with Sming and the TCP environment of that I suggest you start with the Telnet_TCPServer_TCPClient example.
In there you will see how to start tcpserver and telnetserver. And also experiment with debug to telnet.

Second step would then be to start implementing you tcp bridge functionality.
When that is working and finished, you can easily add the rboot functionality.

fvpalha, 2015-10-30:
Thank you @hreintke . I will follow your tips, after I will share the example.

hreintke, 2015-10-30:
@fvpalha : 
OK, if you have trouble with creating the appliaction just leave a question on gitter or forum.

closed 2015-10-30 by hreintke