SmingHub/Sming #381: problem with xtensa-lx106-elf/bin

fenixscd, 2015-10-30:
after instaling developers chocolait Kit, when i compile i get this error.

make all 
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out
c:/espressif/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/5.1.0/../../../../xtensa-lx106-elf/bin/ld.exe: cannot find -lsming
collect2.exe: error: ld returned 1 exit status
make: *** [out/build/app.out] Error 1

is there any possible solution or fix?
i have researched and i think it could be the swich on the version of the program.. but i dont know..

thx and regards.

raburton, 2015-10-30:
How many people are going to report this as an issue? Please read the release notes, you need to build libsming.

closed 2015-10-30 by raburton

kireevco, 2015-10-30:
@raburton people are just simply people :) I was expecting this.

fenixscd, 2015-10-30:
Thanks for the info