SmingHub/Sming #382: Smartconfig API changes

harry-boe, 2015-10-30:
The smartconfig api has changes in SDK 1.4.

The Basic_SmartConfig app uses a local override of the api "smartconfig.h"  that's why it still compiles.
To get it back to work with SDK 1.4 the following changes need to by applied:

application.cpp: 
replace    smartconfig_start(SC_TYPE_ESPTOUCH, smartconfigDone);
with    	smartconfig_set_type(SC_TYPE_ESPTOUCH);
	        smartconfig_start(smartconfigDone);

user_config.h:
replace    #include "smartconfig.h"
	       #include <smartconfig.h>

(got it tested with https://github.com/EspressifApp/EsptouchForIOS)

raburton, 2015-10-30:
I think we need to consider moving to sdk 1.4 and making it a requirement.

fenixscd, 2015-10-30:
I can not find where download the version 1.4

harry-boe, 2015-10-30:
http://bbs.espressif.com/viewtopic.php?f=46&t=1124 

it's a attachment at the end of the post.

fenixscd, 2015-10-30:
Thanks for the info

raburton, 2015-10-31:
Please see PR #389. I've incorporated smartconfig into the WifiStation class now, see the updated sample. Let me know if it works and does what you need it to.

harry-boe, 2015-11-01:
I can see what you did from the discussion in #389 but i can't find the changes in the development branch.
Is there another branch to check out ?

harry-boe, 2015-11-02:
Ok, got it. Test against the raburton/sming sdk14 branch.

Working fine.  Thx

closed 2015-11-02 by harry-boe

hreintke, 2015-11-02:
@harry-boe : 
The "formal way" is getting the PR from Sminghub/Sming.

git fetch origin pull/389/head:smartconf
git checkout smartconf

harry-boe, 2015-11-02:
thanks for the hint