SmingHub/Sming #383: Community Communication

hreintke, 2015-10-30:
@All : I see this issue list getting more a forum character and also in gitter regularly coming questions/remarks which would be good to be held in a more searchable location to prevent multiple times answering the same questions.

Back in August I made the following suggestion which was overwhelmed by the discussion we had on gitflow and no feedback on this.

>Related to this. We now have three communication paths. Chat, Github and Forum.
In all of those we have questions, issues, bugs,... And searching on earlier info is difficult.
Maybe we should get as much as possible to a differentiation of usage.

>    Chat : Used for quick information exchange, questions are expected to be answered/solved within a day. History deleted after one week. Mostly used by frequent users/developers.
>    Forum : Used for Information exchange 1/ for first-time/early user knowledge 2/ larger subjects, enhancement request and possible bugs. Also entry point for new users. History is kept. Well searchable. Probably we need some subforums.
>    Github : Used for registration and follow up on Identified/Verified bugs and enhancement requests. Also for Push Requests.

Now we see an increasing userbase and still scattered information exchange, I think we really should consider moving into the above communication.

Feedback appreciated.

raburton, 2015-10-31:
I certainly agree we need to stop people using the issue tracker as a forum, but I'm not sure how we do that. Perhaps simply more prominent instructions on how to get help on the main page. More documentation, FAQs etc. in the wiki would be a good idea too.

hreintke, 2015-10-31:
We can do that by 
- Agreeing that we will use the the communication as mentioned above.
- Put that on a prominent location in Sming Readme and Wiki
- Push users which use otherwise by directing then to the correct location without directly answering the question

kireevco, 2015-10-31:
I agree that users should be able to get answers to simple and known questions easier. On the other hand I don't think forum on esp8266 will solve this problem. I don't see any problem in using tickets per se, if questions are meaningful. 

The main problem I see is users and human nature. We need to keep that in mind and keep sming flow as simple as possible. That's the only long term solution. The whole flow has to be implemented in such a way where user performs 1 standard action to get and 1 to run example. Even 2 actions is too much nowadays when everyone is used to web :)

I can't accept the theory that sming community are developers and they can handle that. They can't.

Let's not force users to ask questions differently or read additional documentation to simply get started. Let's make sming easier and with sane defaults. This is a key to success.

raburton, 2015-10-31:
>The whole flow has to be implemented in such a way where user performs 1 standard action to get and 1 to run example. Even 2 actions is too much nowadays when everyone is used to web :)
> I can't accept the theory that sming community are developers and they can handle that. They can't.

While I think the sentiment is admirable, I don't share it. Our users are supposed to be writing code in C/C++, if that doesn't make them developers too I don't know what would. If others want to make it more user friendly to these people I don't object, but that's pretty low down my own list of priorities. I'd rather spend the little time I have to work on this project doing things I think are more useful to the project rather than to some of its less able users. Now if making it more user friendly stops the same person opening multiple tickets for the same non-issue then I am in favour of that, but it isn't going to be me that does it I'm afraid.

hreintke, 2015-10-31:
@kireevco : 
There a two items which you should not mix up.
1/ New users who have trouble getting started
2/ Communication between users (new and old) and sming developers

For 1/ the documentation and easy installation process would be beneficial -> that is where your ideas of deployment fit in.
For 2/ The current way of working is absolutely inadequate.
Gitter and Github issues have no way of easy searching -> History and "old questions" are not kept visible and will be asked over and over again. 
Github issues is list is an unsorted/scattered list with user questions and development discussions. 
Esp8266 forum activity will give visibility of Sming to the community which should attract both users and developers. And main : the forum has history and information stays. With some "pinned items" we have a standard way of showing information.
I've asked @alonewolfx2 already some time ago to create a number of subforums (just waiting to be done by main moderator) on the esp8266 to questions/communication from us to the community can be more structured.
Pushing users to use the new maybe will lead to some extra in the beginning but will benefit in a short time.

kireevco, 2015-11-04:
> While I think the sentiment is admirable, I don't share it. Our users are supposed to be writing code in C/C++, if that doesn't make them developers too I don't know what would. If others want to [...]

@raburton We value you for greater things, don't worry! :) 

>For 1/ the documentation and easy installation process would be beneficial -> that is where your ideas of deployment fit in.
ok

>For 2/ The current way of working is absolutely inadequate.
Gitter and Github issues have no way of easy searching -> History and "old questions" are not kept visible and will be asked over and over again. 
Github issues is list is an unsorted/scattered list with user questions and development discussions. 
Esp8266 forum activity will give visibility of Sming to the community which should attract both users and developers.

I still think forcing users to use forum is not a solution by itself. It's just a tool with other means of usage.
I agree github seems scattered right now, but I'm looking at other github projects and a lot of them use solely github issues for all kinds of tickets, we just need a little bit more structure. Github labels should be used, it is a very powerful tool. 

>  And main : the forum has history and information stays. With some "pinned items" we have a standard way of showing information.

Can you give me an example of a pinned item, that can not be placed in README.md or wiki?

Also, one of the reasons why I mixed up these two questions is that users are normally lazy and they search for one first solution (which ends up to be github issues, since they already have github account and already downloaded sming or examples from here). You simple can't force a random user that has an issue search before ask (although you can train). Even with github they can google an error, and github issue will show up first.

@raburton @hreintke you are developers, that's why you contribute to sming core. Please don't overestimate other Sming users. They are close or equal to Arduino (atmel/esp8266) users.

#### What I'm thinking is that:
- We should go along with users natuarl flow and keep using github issues
- Maintain basic documentation as much as possible in wiki/readme. In addition keeping PRs comment section informative (what?/how?/why?) will help us build changelogs and documentation.
- Fix bugs that produce most of user issues (for you and @raburton it might _not_ look like bugs, but they are, because they cause trouble for average user).
- Manage github labels in a better way, so people can always come back to user issues by clicking on a label. For instance, label it as _User_ and provide a link in troubleshooting wiki to tickets with that label.

hreintke, 2015-11-29:
Proposal  not accepted.

closed 2015-11-29 by hreintke