SmingHub/Sming #385: Proposal, sdk v1.4

raburton, 2015-10-31:
I propose for the next release of Sming we move to (and require) SDK v1.4. A number of changes have been made in SDK v1.4 which we need to update our code to use, but these are not all easily backwards compatible. All common build platforms (esp-open-sdk for linux, unofficial esp8266 dev kit for windows) all use v1.4 by default now.

kireevco, 2015-10-31:
@raburton in my opinion it is always good to move forward. And as for me nothing stops us from creating a separate branch (like _feature/sdk1.4_ or so), and make work there. 

What people think?

profra, 2015-10-31:
Hi folks, I installed Sming yesterday evening, everything was functioning at the first time but I tried only Basic_Blink before I went to bed... also the  whole library was build with SDK 1.4 without any erros... as far as I can see, I am total newbie... If I can recommend please switch to new SDK 1.4 .... before the version 1.5 is released :-) .... But what I really miss is better documentation, especially the story about Sming to get the right backroud idea... WHY Sming, WHAT are the parts of Sming, HOW these parts communicate.... So, thank you, Sming seems to be excellent tool for projects with ESP8266!!!

raburton, 2015-10-31:
I agree about documentation, we need some, at the moment if there is no api
documentation you have to look through the many samples as a substitute.
Problem is nobody I'd interested in doing that, I'm certainly not, so
unless someone volunteers I don't expect we'll ever have any.

hreintke, 2015-10-31:
As far as I know the latest UDK Releade 2.0.8 is still on Espressif SDK 1.3.0
However the UDK can be updated manually to use SDK 1.4.0 but that requires an additional step for windows users of Sming.

If we still decide on moving to SDK 1.4.0 into Sming 1.5.0 we should opt for a quick integration into the develop branch. This to have as much of testing/usage with the new SDK.
Whether we do the quick integration with use of PR (like hw_pwm) or feature branch (like rboot) is not a big difference.

For the backward compatibility issues which come with the upgrade we do need them documented in the PR/Branch from the start otherwise we will get a number of questions/remarks which can be prevented.

profra, 2015-10-31:
Yes, you are right, I had to go through some examples to understand the matter... The good documentation is really hard task, I would like to help but I don't have a head  good enough for it and also time... IMO it is task for creators, otherwise Sming cannot be widely spread and used....

raburton, 2015-10-31:
> IMO it is task for creators, otherwise Sming cannot be widely spread and used...

People keep making comments like that and I've had similar related to rBoot. While there are advantages to the project being more popular and more widely used (primarily attracting more developers) I'm not that bothered otherwise. I'm not a salesman and I don't make any money from people using this, if people want to use it then fine, but if they don't, if they find something else easier, then good for them. My intention in joining the Sming project was to help make it better for my own use, not for anyone else's or to encourage others to use it. That doesn't mean if people come to me with requests for help with things or changes or new features, etc, I won't consider them because I do, but that's because I think it's a fun challenge or I want it myself or because I'm being nice - but my goal is not to make Sming spread or be more popular. Documentation certainly is a good idea, but I'm not sure you'll find many people here amongst the "creators" (most of us actually joined after the core was created, although it's constantly changing) that wants to do this themselves or it would have already been done.

raburton, 2015-10-31:
pr #389 now has the malloc related changes in it, i'll look at other issues as people post them here...

robotiko, 2015-10-31:
I do support the SDK 1.4.0 >=  idea  for Sming and forget about backwards compatibility for earlier Sming versions.
Keep SDK compatibility would introduce a lot of troubles for core development that were never addressed before either. Based on the gitter conversations, most developers always try with latest SDK and update by default (because  project doc just points to aged SDK compatibility) so looks like wont be a big concern anyway. 
IMHO The Sming and SDK low maturity at this point don't justify the headaches involved in backports and community is not using this with long life support concern.

johndoe8967, 2015-10-31:
I also agree on moving on to the newest SDK.
But just keep in mind that with the newer SDK the binary als always grows in size.
I have one project with about 251kb (SDK 1.3) and 260kb (SDK 1.4) 
So some projects might not fit into the flash and you have to stay with the older SDK.

raburton, 2015-10-31:
People will have to stay with older versions of Sming, or backport newer features if they need. We can't easily support older versions because espressif don't include an sdk version number in their headers which you could use to make code conditional. Espressif need a lot of help with their sdk, but at least they provide one!

tprochazka, 2015-11-06:
The title of the issues should be changed from 1.4 to 1.5

raburton, 2015-11-08:
> The title of the issues should be changed from 1.4 to 1.5

No it shouldn't.

piperpilot, 2015-11-19:
Any update on merging this one to develop?

raburton, 2015-11-19:
It was voted down by windows users as their most common development environment doesn't yet include SDK v1.4. The work is done though, see PR #389.

raburton, 2015-11-22:
Now merged by popular support. @kireevco travis doesn't like, I think it only uses a v1.2 sdk.

closed 2015-11-22 by raburton