SmingHub/Sming #387: SPIFF_SIZE evaluation produce bigger output than SPI SIZE?

tprochazka, 2015-10-31:
I tested latest Sming on my ESP-02 with the program using SPIFF.
And it completely doesn't work, program was not started.

It generate these files
```
0x00000.bin	32 368	
0x09000.bin	259 130	
spiff_rom.bin	262 144	
```

So the whole size is bigger than 512kB flash on ESP-02
It is right?

According this code in make file

```
# flash larger than 1024KB only use 1024KB to storage user1.bin and user2.bin
ifeq ($(SPI_SIZE), 256K)
	flashimageoptions += -fs 2m
	SPIFF_SIZE ?= 131072  #128K
else
    ifeq ($(SPI_SIZE), 1M)
	flashimageoptions += -fs 8m
	SPIFF_SIZE ?= 524288  #512K
    else
        ifeq ($(SPI_SIZE), 2M)
		flashimageoptions += -fs 16m
		SPIFF_SIZE ?= 524288  #512K
        else
            ifeq ($(SPI_SIZE), 4M)
			flashimageoptions += -fs 32m
			SPIFF_SIZE ?= 524288  #512K
            else
			flashimageoptions += -fs 4m
			SPIFF_SIZE ?= 262144  #256K
            endif
        endif
    endif
endif
```

the of spiffy is really 262144 .
Maybe because the espressif SDK is now bigger than was?
Is possible evaluate SPIFFY size according others .bin files?

I tried decrease 262144  to 196608 and my code works now.

And btw.  Why you don't use this syntax for multiple if in makefile
```
# flash larger than 1024KB only use 1024KB to storage user1.bin and user2.bin
ifeq ($(SPI_SIZE), 256K)
	flashimageoptions += -fs 2m
	SPIFF_SIZE ?= 131072  #128K
else ifeq ($(SPI_SIZE), 1M)
	flashimageoptions += -fs 8m
	SPIFF_SIZE ?= 524288  #512K
else ifeq ($(SPI_SIZE), 2M)
	flashimageoptions += -fs 16m
	SPIFF_SIZE ?= 524288  #512K
else ifeq ($(SPI_SIZE), 4M)
	lashimageoptions += -fs 32m
	SPIFF_SIZE ?= 524288  #512K
else
	flashimageoptions += -fs 4m
	SPIFF_SIZE ?= 262144  #256K
endif
```

raburton, 2015-10-31:
What size did you set your SPIFF_SIZE variable to?

tprochazka, 2015-10-31:
I'm not sure that understand your question.
By default I didn't set any size to SPIFF_SIZE, because there are auto evaluation in the project makefile.
But When I found that it doesn't work I set it to the 192kB manually. But I mentioned that in the original post.

raburton, 2015-10-31:
Sorry, my mistake, I meant what did you set SPI_SIZE to? That is what is used for automatic calculation of SPIFF_SIZE (as you can see from the makefile).

tprochazka, 2015-10-31:
I set 512K as SPI_SIZE

raburton, 2015-10-31:
Hmm, yes, i see the problem then.
It might be in part that sming itself has got bigger, or it could be that your app is large itself, either way the default is isn't going to work for you. Are you just using a simple sample or a bigger app of your own?

I personally think the auto size and position stuff is bad, it isn't (and can't be) clever enough to be safe. My advice is to always specify SPI_SIZE manually, and we should remove this auto size stuff. In the mean time we could lower the default on the 512k code.

tprochazka, 2015-11-01:
I'm totally newbie in memory management/mapping on ESP8266 and it's quite complicated I think, some more advanced advisor/configurator for the beginners would be great. Or some great article about it, at least.

Because I agree that manually chosen SPIFF_SIZE is better, but it is hard to find out it. For example I'm looking for the way how to keep SPIFF partition unchanged during reflash.

tprochazka, 2015-11-01:
And yes. I have only combined 3 examples together, nothing complicated. Program that ismply use smart config, mDSN, ftperver, http server which control one LED via REST API.

raburton, 2015-11-01:
Perhaps we need to decrease the default size when using small flash sizes then. Anyone else have comments on this?

closed 2015-11-02 by raburton

hreintke, 2015-11-02:
@raburton : 
Isn't the same update necessary for makefile-project ?
If so, should we take an action to combine the two ?

raburton, 2015-11-03:
@hreintke I did it in both, see the commit.

commit 6d26f299 (2015-11-02): reduce default spiffs size for 512k flash, closes #387