SmingHub/Sming #388: When installing with Chocolatey the development environment does not work, compile error.

fenixscd, 2015-10-31:
When installing with Chocolatey the development environment does not work, compile error.

C + app / application.cpp
AR out / build / app_app.a
LD out / build / app.out
c:/espressif/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/5.1.0/../../../../xtensa-lx106-elf/bin/ld.exe: can not find -lsming
collect2.exe: error: ld returned 1 exit status
make: *** [out / build / app.out] Error 1

raburton, 2015-10-31:
please see your own issue #381, from just yesterday, in which you had the same problem and the answer was given

raburton, 2015-10-31:
in addition, please only use the github issue tracker for reporting bugs or feature requests, it is not a user help forum

closed 2015-10-31 by raburton

fenixscd, 2015-10-31:
ok