SmingHub/Sming #391: What are the lwip includes for?

raburton, 2015-10-31:
Found an issue when moving to sdk v1.4 that was is system/include/lwip/mem.h but I can't work out why we have all these includes at all! We don't use the libopenlwip that is included in sming so I've removed it, but we seem to use these headers and I don't understand why.

hreintke, 2015-10-31:
In early days sming was using the openlwip library.
When connecting issues came when upgrading to a new SDK it was changed to the espressif lwip.
Do you know where the headers are used ?

raburton, 2015-10-31:
Delete the folder and try to compile it, you'll get a big list of places!

raburton, 2015-10-31:
It looks like we use structures from those headers that don't exist in the Espressif headers, which is worrying.

hreintke, 2015-10-31:
Can't check now but will tomorrow.
But I suppose it is because Sming doesn't use the "espressif network layer" but directly the underlying LWIP network layer. That is not included in the espressif headers.

raburton, 2015-10-31:
Wow, no wonder we have network problems then!

hreintke, 2015-10-31:
Of course not absolutely  sure but LWIP is more open/standard than the espressif layer above it.
But.. Since a few weeks there is an "Open source LWIP for ESP_IOT_SDK_V1.4.0" from espressif.
http://bbs.espressif.com/viewtopic.php?f=46&t=1221
Somewhere on the todo list is a check whether/how we should migrate to that.

raburton, 2015-10-31:
I'm looking at that at the moment, but as usual from espressif there is no information about it at all, just a file dumped in a forum.

hreintke, 2015-10-31:
I know, and as it at the base of Sming we should be very careful on this and not updating without severe testing.

raburton, 2015-10-31:
Don't worry, I can't even work out how you are supposed to compile it.

raburton, 2015-11-01:
Well got some stuff compiling, but not had any time to play with it further. Not sure how much time I'll have over the next week either, got night shifts this week and some house rewiring to do in between.

hreintke, 2015-11-02:
@raburton : 
I don't think this id the way to go.
@kireevco is busy having the esp-open-sdk available on all platforms.
According the documentation that includes `lwIP, http://savannah.nongnu.org/projects/lwip/ `
Wouldn't that remove the necessity to include espressif lwip ?

@kireevco : Please confirm

raburton, 2015-11-02:
I assume the espressif libraries use the function in their version of lwip, so if we go with a pure lwip I don't imagine their libraries will link against it. I'm assuming the new one they have released is normal lwip with their bits on top, but I haven't looked into it that much yet.

hreintke, 2015-12-09:
@raburton : 
In porting Sming to RTOS I ran into a LWIP issue. 
Did you get the Open source LWIP for ESP_IOT_SDK_ compiling ?
Maybe I can use (parts) of that to fix or bugfinding.

raburton, 2015-12-09:
I got as far as compiling it, but I didn't go on to try and use it because you told me not to!

hreintke, 2015-12-09:
I know, just asking what you did to make it compile, not to finish the integration.
Not sure whether I can make that for use in the RTOS SDK but want to give it a try.

There is the directory : lwip_open_src_template_proj, which includes an application user_main.c
Did you compile the application "separate" or or did you compile lwip as library for use ?
Did you copy the contents of lwip_open_src_template_proj in the ESP8266_SDK directory ?
Which changes did you make to makefile(s).

raburton, 2015-12-10:
I added my own make files to build the libraries. Try this: https://dl.dropboxusercontent.com/u/5500141/makes.zip

hreintke, 2015-12-10:
Thx, Will give it a try later today.

hreintke, 2015-12-10:
@raburton : 
Got the compile working. Needed to include the sdk_incdir in the cc line : 
`@$(CC) -I. -I../../../include $(SDK_INCDIR) $(CFLAGS) -o $@ -c $<`

Changing to the RTOS SDK does not work however.
Continuing investigation.

closed 2015-12-21 by hreintke