SmingHub/Sming #392: weird characters on Serial port, and no other serial output

2015mp, 2015-11-01:
I have a NodeMCU,  and when I send any including the serial samples, I get the output below

rll��|�l�|�l�b|����r�b�b��nn�lnn���bp�lrlrlp�n��lbn�|��ll��b��nn�l��l`�nnl�l`nr���n��``�n��b�nl���nn��lp�n�r�������bn�|l�ll��r�b��nn�l�l`rll��|�l�|�l�b|����r�b�b��nn�lnn���bp�lrlrlp�n��lbn�|��ll��b��nn�l��l`�nnl�l`nr���n��``�n��b�nl���nn��lp�n�r�������bn�|l�ll��r�b��nn�l�l`rll��|�l�|�l�b|����r�b�b��nn�lnn���bp�lrlrlp�n��lbn�|��ll��b��nn�l��l`�nnl�l`nr���n��``�n��b�nl���nn��lp�n�r�������bn�|l�ll��r�b��nn�l�l`

hreintke, 2015-11-01:
Did you check baudrate ?

2015mp, 2015-11-02:
Thanks! I found the issue, I had a nodeMCU that uses flash that needs to be programmed in DIO.  The default for the esptool.py script was to program in QIO.  

To anybody interested in the steps I took to find it
The debug steps were to to look at the output at 74880 baud, then see the error message.  That pointed me to a thread
http://bbs.espressif.com/viewtopic.php?t=25
that had details on the different error messages in the boot area.
so once I googled the flash ID, and got the data sheet, it mentioned that this could be programmed DIO/QIO.  Tried DIO, and ESP8266 came alive :)

closed 2015-11-04 by kireevco