SmingHub/Sming #393: `.irom0.text' will not fit in region `irom0_0_seg'

tprochazka, 2015-11-01:
I got this error message "`.irom0.text' will not fit in region `irom0_0_seg'"
I found here possible solution
http://bbs.espressif.com/viewtopic.php?f=61&t=166

Sming use this ld file sming\Sming\compiler\ld\eagle.app.v6.cpp.ld
I compared it with files in the Espressif SDK and it is quite different.
I manually increased irom0_0_seg len value and it looks that it works now.

It is good idea to use only one ld file or there should be more and decision should be according defined SPI_SIZE?

As I already wrote in the another issue. There should be some smart tool that is able to configure or advise right memory configuration especially for new users. It is quite complicated.

My application just combine several Sming examples, nothing complicated.

raburton, 2015-11-01:
Increasing irom0_0_seg is the correct thing to do if you need more space.

You can't pass variables to the linker file, so there is no easy way to do this automatically based on SPI_SIZE. You'd have to copy the file and use tools to edit the file from the command line (not very platform independent).

closed 2015-11-01 by raburton

tprochazka, 2015-11-03:
It is not possible to increase it automatically according code size?

raburton, 2015-11-03:
No it isn't.

robotiko, 2015-11-30:
@raburton 
I just updated to latest dev branch (updated sdk from 1.3 to 1.4 )and rebuild my project and got this error. 
I bet the new Sming/ sdk code is heavier.
What are reasonable values to set there?

I'm currently using rboot in my project , this means the changes should be just done in rom0.ld? or also in the Smign one?

Thanks

raburton, 2015-11-30:
Depends what size rom you have, how many roms you are ysing and how they are laid out. If you have 1mb per rom you can set the size up to 0xF0000 comfortably.