SmingHub/Sming #394: Sming versioning

flexiti, 2015-11-01:
I suggest the introduction of the version number in the format X.X.XX  eg. 1.4.00 , 1.4.01 and so on
This will allow us to synchronize numbering with the numbering SDK.
If, for example. The SDK is 1.4, a version Sming varies from 1.4.00 to 1.4.99
If the version of the SDK will change to 1.5 is the first version Sming will have the number 1.5.00

Following this proposal, a new version Sming for today  should have  (have now 1.4.0) a number eg. 1.4.01
or if we like 1.4.10 (because it is major change from SDK 1.3 to SDK 1.4)

ps.We can also use X.X.X format eg 1.4.0,  1.4.1 with with a smaller number of changes between changes SDK
With this solution a new version sming should be numbered 1.4.1, unless changes SDK to 1.5, then 1.5.0

piperpilot, 2015-11-03:
+1, I think this makes a lot of sense and will aid in understanding what versions are required.

hreintke, 2015-11-03:
I am not sure.
First : We don't know what the future SDK numbering will be. That is outside our hands.
And : Usual verison numbering is like the one used now 1.3.0 -> 1.4.0 -> 1.5.0 etc
The specific situation now is that the SDK release is almost the same as the Sming release.

If we agree that at least every SDK upgrade release gets a "major release status" we have
Sming 2.0.0 with SDK 1.4.0 -> 2.1.0 with SDK 1.4.0
Sming 3.0.0 with SDK 1.5.0 etc
When Large impact functionality comes we can to f.e.
Sming 4.0.0 with SDK 1.5.0 + added functionality
Sming 5.0.0 with SDK 1.6.0

kireevco, 2015-11-04:
Not a good idea, i'm with @hreintke. 
The issue that we will be locked to only one number in X.X.YY, which is YY. And we don't know that Espressiff won't use it in the future.

We need to stick to it's own [semantic versioning](http://semver.org/) in sming and follow _our_ release cycle. We can explicitly specify what sdk version a specific sming release is using.

flexiti, 2015-11-04:
It's just going to have available SDK 1.6, and at that time will be available Sming 1.5 working with SDK 1.4 - terrible confusion and completely do not know why you're afraid what number will be SDK - then sming will be able to have the same

I do not know why You not like this change :) 

ps.
If the Sming changes are every month and the sdk does not change, this 99 change means eight years - with such a technical advance no one will remember ESP8266 :)

kireevco, 2015-11-04:
@flexiti 
I'm not sure who are you replying to, but I assume to my message about X.X.YY
>It's just going to have available SDK 1.6, and at that time will be available Sming 1.5 working with SDK 1.4 - terrible confusion and completely do not know why you're afraid what number will be SDK - then sming will be able to have the same

- What if they will have 1.6.4 or 1.6.90?
- What if they will send us a patch for 1.5.0, which version sming will have?

I'll give you an example why. If you look at Java world, you'll see that very little amount of jvm projects use jvm version as a base version. Hundreds of big jvm projects have their own versions.

And to make everyone's lifes easier eventually sming will contain it's own espressif sdk as a part of it, so no version matching will be required. You download 'sming-full-1.9.3.zip' and point your eclipse to it. How does that sound?

flexiti, 2015-11-04:
@kireevco 
As someone does not want do always find problems

ok, let it be as it is

closed 2015-11-08 by raburton