SmingHub/Sming #395: warning: cannot find entry symbol call_user_start

Picatek, 2015-11-02:
Installed the latest(2015-11-02) sming on windows (no errors).
Build the  SmingFramework (no error)

However when try to build the Basic_Blink example project I got first:
warning: cannot find entry symbol call_user_start; defaulting to 40100000

and then (see below):
KeyError: 'call_user_start'

I also tried other projects(Basic_Serial) with the same errors
Any idea ?

10:44:31 **** Build of configuration Sming for project Basic_Blink ****
make all 
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out
c:/espressif/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/5.1.0/../../../../xtensa-lx106-elf/bin/ld.exe: warning: cannot find entry symbol call_user_start; defaulting to 40100000

#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8000|       0
    rodata|           ReadOnly Data (RAM)|    3FFE8000|    3FFE8008|       8
       bss|      Uninitialized Data (RAM)|    3FFE8008|    3FFE8008|       0
      text|            Cached Code (IRAM)|    40100000|    40100030|      48
irom0_text|           Uncached Code (SPI)|    40209000|    40209000|       0
Total Used RAM : 8
Free RAM : 81912
Free IRam : 32738
------------------------------------------------------------------------------
# Generating image...
Traceback (most recent call last):
  File "esptool.py", line 675, in <module>
  File "esptool.py", line 439, in get_symbol_addr
KeyError: 'call_user_start'
make: *** [out/build/app.out] Error 255

10:44:32 Build Finished (took 1s.100ms)

kireevco, 2015-11-04:
@Picatek Please provide debug info from the Wiki?

hreintke, 2015-11-14:
Is his related to #402 ?

hreintke, 2015-11-15:
@Picatek @vsirvent :
This might be an esptool issue, especially on 64 bit windows.
Other tests revealed that UDK206 has an "older" version of esptool than UDK208. 
This is confirmed by the release notes from UDK. -> In these tests, using UDK208 worked OK.

Can you test whether your "erroneous behavior" still exists when using UDK208 and report back ?

hreintke, 2015-11-29:
solution discussed in  #447 and #450

closed 2015-11-29 by hreintke