SmingHub/Sming #396: user_config.h: No such file or directory'

vnvnsairam, 2015-11-02:
19:33:46 **** Incremental Build of configuration Sming for project LED_WS2812 ****
make all 
Building file: ../app/application.cpp
Invoking: GCC C++ Compiler
g++ -I"C:\tools\sming\Sming" -I"C:\tools\sming\Sming/system/include" -I"C:\tools\sming\Sming/Libraries" -I"c:/Espressif/sdk/include" -O2 -g -Wall -c -fmessage-length=0 -MMD -MP -MF"app/application.d" -MT"app/application.d" -o "app/application.o" "../app/application.cpp"
../app/application.cpp:1:25: fatal error: user_config.h: No such file or directory
 #include <user_config.h>
                         ^
compilation terminated.
make: *** [app/application.o] Error 1

19:33:46 Build Finished (took 214ms)

file is present in includes folder can any one explain this ?

kireevco, 2015-11-04:
@vnvnsairam did you go through https://github.com/SmingHub/Sming/wiki/Troubleshooting-Windows? Please provide debug output.

vnvnsairam, 2015-11-04:
@kireevco  thanks i found the solution...im facing this error with BMP180 example
Could not connect to BMP180.
Initialize
AC1:	0
AC2:	0
AC3:	0
AC4:	0
AC5:	0
AC6:	0
B1:	0
B2:	0
MB:	0
MC:	0
MD:	0
cal
Start readingFatal exception (0):
epc1=0x4000dce5, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 21400, room 16 
tail 8
chksum 0x97
load 0x3ffe8000, len 848, room 0 
tail 0
chksum 0xba
load 0x3ffe8350, len 1864, room 8 
tail 0
chksum 0xc7
csum 0xc7
rl

vnvnsairam, 2015-11-04:
reinstall worked for me

closed 2015-11-04 by vnvnsairam