SmingHub/Sming #397: GPIO9 and 10 is overwritten during  Adafruit_SSD1306:display() method

tprochazka, 2015-11-03:
If you will use GPIO9 a 10 as output pins together with Adafruit_SSD1306 library, state of these GPIOS will be overwritten every time that OLED is updated by display() method.

I'm not sure just now with the exact reason now.
But I'm very close. I tried to replace 
```
*csport |= cspinmask;
*dcport |= dcpinmask;
*csport &= ~cspinmask;
*csport |= cspinmask;
```
by
```
digitalWrite(cs, HIGH);
digitalWrite(dc, HIGH);
digitalWrite(cs, LOW);
digitalWrite(cs, HIGH);
```
In Adafruit_SSD1306.cpp.
And problem is gone!

It looks that bit mask is incorrect and change more pins!
I checked that uint8_t is used for PortReg and PortMask it is correct for ESP like 32 bit CPU? 
Or it is problem in digitalPinToBitMask() or portOutputRegister() ?

tprochazka, 2015-11-03:
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Arduino.h#L227

anakod, 2015-11-09:
Ok, got it. I think we should make minimal test code for repeating this problem and when we can start working on register implementation part.

tprochazka, 2015-11-10:
I already localize and found the problem, 8 bit variable was used in Sming instead of 32 bit for GPIO register.

anakod, 2015-11-10:
Please look my comment to related P-R. I think it isn't right and willn't work for high pin numbers.

tprochazka, 2015-11-14:
@anakod  ESP8266 Arduino project use also 32 bit register, why do you think that it will not work?

hreintke, 2016-05-23:
Issue should be moved to SmingRTOS.

closed 2016-05-23 by hreintke

tprochazka, 2017-07-14:
I don't understand why this was fixed in RTOS version and not here, id this is still under development, or not?