SmingHub/Sming #399: I2C issue while reading from HTU21DF

hmms, 2015-11-04:
i was trying to read the HTU21DF using the Adafruit library (https://github.com/adafruit/Adafruit_HTU21DF_Library) the 'begin' function works and the chip is detected, but the data requested using the Wire.requestFrom() doesnt seem to work and returns zero bytes, also Wire.Available() doesn't seem to work and returns zero. are there any work arounds? I would be glad to contribute to help solve this issue, pointers appreciated!

avr39-ripe, 2015-11-04:
I think I2C bug WAS resolved by this commit https://github.com/SmingHub/Sming/commit/eb95d30760a02412be4da8bfd3051557916ea6f9

What version of Sming do you use? Try to update to latest release. If issue remains - feel free to ask for help here!

comino, 2015-11-04:
I have the same issue with a BMP180 - returns only zeros.

flexiti, 2015-11-04:
BMP 180 works ok on Esp12 end esp01 , I am using develop version of Sming (2 weeks old).

hmms, 2015-11-04:
@avr39-ripe I think i am using an older build as you suspected, I will give it a try tonight with eb95d30 and will let you know of the results.

hmms, 2015-11-05:
I tried with the latest build( afe5393 ) Issue still persists. I dont have a DSO or a scope to see exactly whats going on but a friend of mine has a Logic analyzer will borrow that and check what might be going wrong. Thanks again for the help :)

avr39-ripe, 2015-11-05:
One more thought: may be you use wrong I2C pins? Are you sure, that you connect your device to right pins? Try to specify pins directly before Wire.begin() in init() syntax is void pins(int pinSCL, int pinSDA); // Can be called only before begin():
``` C++
        Wire.pins(0, 2);
        Wire.begin();
```

comino, 2015-11-05:
I also still have the problem. BMP180 works with Arduino dode. Smite BMP180 example with newest version is not working. Im using the correct pins, also tried "Wire.pins(0,2)" and a lot of other stuff. I dont get whats wrong. I only get zeros what-ever I do. 

Increased the delay between pulling up the lines and starting the transmisson. No difference.

avr39-ripe, 2015-11-05:
@comino by "Arduino code" you mean real AVR-based arduino+its code, or esp8266-based Arduino software platform?
Still can't understand, why you have problems. I use Sming + DS3231 RTC and Sming + I2C based LCD (via PCF8574 port expander) without any problems.

flexiti, 2015-11-05:
@comino 

Try:

        BMP180 barometer;
	Wire.pins(0,2);  // I am using (12,13)
	Wire.begin();
	if(!barometer.EnsureConnected())
		Serial.println("Could not connect to BMP180.");
        barometer.Initialize();
	long currentPressure = barometer.GetPressure();

	Serial.print("Pressure: ");
	Serial.print(currentPressure);
	Serial.print(" Pa");     

Maybe You switched ports ?? 0 <-> 2
Some board BMP180 require a 5V power supply (they have their own power supply 3v3), Have  you checked yours?

comino, 2015-11-05:
ESP8266 Arduino + same board its working. Checked several times if BMP still works on adurino - works perfect. Tried what you menitoned @flexiti but still no success. But since its working for you guys, the problem must either be on my computer or sitting in front of it and not in Sming ;). Thanks for your support.

wal-stefan, 2015-11-05:
Here is *my* complete initialization sequence. Maybe, not everything is needed, but it is working fine for me. I am using the default pins. In this case, remember to remove the "flash enable" jumper after flashing - at least on ESP-12 it shares the SCL pin of he I2C-bus.

````
BMP180 barometer;

void initBarometer()
{
  Wire.begin();
  if(!barometer.EnsureConnected()) {
    LOG("Could not connect to BMP180");
    return;
  }

  LOG("Connected to BMP180");
  barometer.SoftReset();
  barometer.Initialize();
  barometer.PrintCalibrationData();
}
```

hmms, 2015-11-09:
I got the logic analyzer and here's the results, the sensor works fine, the screen grab shows the "read temperate" command(0x40, the chip I2C ID followed by 0xE3) being sent out. the adafruit library works fine with the uno, but when i try it with Sming it fails to read data from the HTU21DF sensor. could it be something with the SCL line?
[image]

tprochazka, 2015-11-09:
I personally think that grab the SPI and I2C implementation from the Arduino ESP8266 project would be best way how to fix a lot of issues in quite small time, they will fix issues instead of us. Arduiono ESP8266 project has much more programmers on the project. But I personally like network API on Sming much more, so I hope that Sming doesn't die.

hmms, 2015-11-25:
@comino I tried the BMP180, and it seems to work, the issue might be in the Adafruit library(i suspect the delay and timing function maybe), The HTU21DF works fine with Arduino for the ESP8266, just not with sming.

alonewolfx2, 2015-11-26:
Bmp180 example working fine for me but I can't read data from attiny85 on i2cbus too.

hreintke, 2016-01-27:
closed due to low activity

closed 2016-01-27 by hreintke